platformio/platform-teensy #1: noteFrequency analysis in audio lib doesn't work in pio

cosmikwolf, 2016-08-22:
I am running into an issue where I have a sketch that works just fine in the regular arduino environment, but does not work when compiled and uploaded in the PlatformIO environment. I manually replaced the framework-arduinoteensy folder with the one I used when successfully testing in the regular arduino.app environment. 

Here is the code:
`#include <Audio.h>
#include <SPI.h>
#include <Wire.h>

AudioInputAnalog              adc(A7);
AudioAnalyzeNoteFrequency     notefreq;
AudioConnection               patchCord1(adc, 0, notefreq, 0);

void setup() {
  AudioMemory(30);

  Serial.begin(9600);

  delay(1000);
  Serial.println("setup complete");
  delay(500);
  pinMode(13, OUTPUT);
  notefreq.begin(.15);
}

void loop() {
  digitalWrite(13, HIGH);
  delay(100);
  digitalWrite(13, LOW);
  delay(100);
  Serial.println(String( millis() ) + "\t");

  if (notefreq.available()) {
    float note = notefreq.read();
    float prob = notefreq.probability();
    Serial.printf("Note: %3.2f | Probability: %.2f\n", note, prob);
  }

}`

You don't actually need to hook any signal up to test. When I run in the arduino environment, everything is working and the LED is blinking. When running in the platform IO environment, the device doesn't get past the notefreq.begin() stage, and does not begin to blink the LED, and then spits about a bunch of garbage into the serial port.

The teensy is running at 96mhz. Is there anything you would have me do to further test this? This happened in the latest version of platform 2.x. Could not test with PIO 3.0 as i could not get 3.0.0a6 to compile my project without other errors (will open bug for that too)

Thanks!

valeros, 2016-08-23:
Hi @cosmikwolf !
Should be fixed in the development branch. Please follow next steps:
```
# uninstall teensy platform
platformio platform uninstall teensy

# uninstall existing PlatformIO
pip uninstall platformio

# install the latest development version
pip install -U https://github.com/platformio/platformio/archive/develop.zip

# install teensy from development branch
platformio platform install https://github.com/platformio/platform-teensy/archive/develop.zip
```

cosmikwolf, 2016-08-27:
Well, there is a bit of progress! 

Now it will get past `  notefreq.begin(.15);` and begin looping, but as soon as I connect an audio rate signal to pin A7, it crashes and prints something like this:

`````
18739
18939
19139
19339
19540
19740
19940
Note:  | Probability: 
1994setup complete    Note:  | Probability:                                                                                                                            g%퍓??5?#׀Zp??Θ??ΚH?????Κ?>????'?<?:?<x`]񳹅>?Hn!4???L?#Lߕ??.????H??? ?                  ?
2123                                        ????O?h??h??????l???~ d????????Od???~ ?~ ??`~ ?~ d???~ ????O?`~ ?~ ????O?d???s?\~ l????#?#433|(???:
````

cosmikwolf, 2016-08-27:
Here is a sketch you can use to test without having to wire anything up. It works just fine in the Arduino IDE, but not in the beta branches you specified above:

````
#include <Audio.h>
#include <SPI.h>
#include <Wire.h>

AudioOutputAnalog             dac;
AudioSynthWaveformSine        sine;
AudioAnalyzeNoteFrequency     notefreq;
AudioConnection               patchCord1(sine, 0, notefreq, 0);

void setup() {
  AudioMemory(30);
  sine.amplitude(1.0);
  sine.frequency(2000);
  notefreq.begin(.15);

  Serial.begin(9600);

  pinMode(13, OUTPUT);

}

void loop() {
  digitalWrite(13, HIGH);
  delay(100);
  digitalWrite(13, LOW);
  delay(100);
  Serial.println(String( millis() ) + "\t");

  if (notefreq.available()) {
    float note = notefreq.read();
    float prob = notefreq.probability();
    Serial.printf("Note: %3.2f | Probability: %.2f\n", note, prob);
  }

}
````

ivankravets, 2016-09-15:
@cosmikwolf could you try new version of development platform `1.1.0`?

```
platformio update
```

cosmikwolf, 2016-09-16:
I get the following error when attempting to compile @ 1.1.0 after the update: 

`--------------------------------------------------------------------------------
AttributeError: 'SConsEnvironment' object has no attribute 'VerboseAction':
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 125:
env.SConscript("$BUILD_SCRIPT")
File "/Users/tenkai/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:

return _SConscript(self.fs, *files, **subst_kw)
File "/Users/tenkai/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/Users/tenkai/.platformio/platforms/teensy/builder/main.py", line 150:
action=env.VerboseAction(" ".join([`

cosmikwolf, 2016-09-16:
I had an older version of pio, so I did an upgrade and now I am at 3.1.0a.

I uninstalled and re-installed teensy 1.1.0 and re-installed and I get this error when compiling now, (worked fine before)

```
/Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/control_ak4558.cpp:9:18: fatal error: Wire.h: No such file or directory
#include "Wire.h"
^

compilation terminated.
Compiling .pioenvs/teensy31/lib/Audio/control_sgtl5000.o
/Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/control_cs4272.cpp:31:18: fatal error: Wire.h: No such file or directory
#include "Wire.h"
^
compilation terminated.
Compiling .pioenvs/teensy31/lib/Audio/control_wm8731.o
/Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/control_sgtl5000.cpp:28:18: fatal error: Wire.h: No such file or directory
#include "Wire.h"
^
compilation terminated.
Compiling .pioenvs/teensy31/lib/Audio/data_ulaw.o
/Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/control_wm8731.cpp:28:18: fatal error: Wire.h: No such file or directory
#include "Wire.h"
^
compilation terminated.
Compiling .pioenvs/teensy31/lib/Audio/data_waveforms.o
Compiling .pioenvs/teensy31/lib/Audio/data_windows.o
Compiling .pioenvs/teensy31/lib/Audio/effect_bitcrusher.o
*** [.pioenvs/teensy31/lib/Audio/control_ak4558.o] Error 1
*** [.pioenvs/teensy31/lib/Audio/control_cs4272.o] Error 1
*** [.pioenvs/teensy31/lib/Audio/control_sgtl5000.o] Error 1
*** [.pioenvs/teensy31/lib/Audio/control_wm8731.o] Error 1
```

Wire.h and the Audio library are both libraries that come with the framework-teensy. The Wire.h lib is present, but its not picking it up for some reason...

cosmikwolf, 2016-09-16:
Is there a way to revert to the previous version so I can keep working in the time being? I now can't compile at all and can't work on my software...

cosmikwolf, 2016-09-16:
After uninstalling pio and re-installing (I had v 3.0.0aSomething) to 3.1.0a it is all working.

ivankravets, 2016-09-16:
@cosmikwolf can I close this issue?

cosmikwolf, 2016-09-26:
yes, you can close the issue now. thank you

closed 2016-09-26 by ivankravets