platformio/platform-teensy #2: Update framework-arduinoteensy

blackketter, 2016-08-22:
Teensyduino has been updated to 1.29:

https://github.com/PaulStoffregen/cores/tree/1.29

Thanks!

valeros, 2016-08-23:
Updated only for PlatformIO 3.0. In order to avoid some issues with libraries, I recommend updating to the latest `teensy` platform and PlatformIO from development branches.
```
# uninstall teensy platform
platformio platform uninstall teensy

# uninstall existing PlatformIO
pip uninstall platformio

# install the latest development version
pip install -U https://github.com/platformio/platformio/archive/develop.zip

# install teensy from development branch
platformio platform install https://github.com/platformio/platform-teensy/archive/develop.zip
```

valeros, 2016-09-02:
Please reopen this issue if you have any problem.

closed 2016-09-02 by valeros

profra, 2016-09-08:
Teensyduino has been updated to 1.30:

https://github.com/PaulStoffregen/cores/tree/1.30

Please and thanks!

ivankravets, 2016-09-08:
@profra please open new issue