platformio/platform-teensy #3: Errors when compiling a teensy audio library project

cosmikwolf, 2016-08-23:
I think all you need to make this happen is include <Audio.h>

```
In file included from /Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/src/Audio.h:16:0,
from /Users/tenkai/Desktop/Projects/flxs1/src/flxs1.ino:1:
/Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/src/DAC.h:21:16: error: expected ')' before '*' token
DACClass(Dacc *_dac, uint32_t _dacId, IRQn_Type _isrId) :
```

valeros, 2016-08-23:
Should be fixed in the development branch. Please follow next steps:
```
# uninstall teensy platform
platformio platform uninstall teensy

# uninstall existing PlatformIO
pip uninstall platformio

# install the latest development version
pip install -U https://github.com/platformio/platformio/archive/develop.zip

# install teensy from development branch
platformio platform install https://github.com/platformio/platform-teensy/archive/develop.zip
```

ivankravets, 2016-08-29:
@cosmikwolf Could you test it?

ivankravets, 2016-09-09:
Please reopen if you still have problem

closed 2016-09-09 by ivankravets