platformio/platform-teensy #4: Add support for new Teensy 3.5 and 3.6

crteensy, 2016-08-31:
There's a pair of new Teensys: https://forum.pjrc.com/threads/34808-K66-Beta-Test

It's not obvious from the first post that there are actually two different new boards, but the differences are outlined in this google spread sheet (not mine): https://docs.google.com/spreadsheets/d/1LSi0c17iqtvpKuNSYksMG306_FpWdJcniSRR6aGNNYQ/edit#gid=95008987

teensyduino 1.30 (https://forum.pjrc.com/threads/36165-Teensyduino-1-30-Beta-3-Available) adds support for the boards to the arduino IDE, so all the necessary information should be available.

I just wanted to try platformio and have no experience with it, so I'm a bit hesitant to try adding support for those.

ivankravets, 2016-09-09:
@crteensy what is your OS?

Please test development version:
```
platformio platform uninstall teensy
platformio platform install https://github.com/platformio/platform-teensy.git
```

Does it work with new Teensy?

closed 2016-09-09 by ivankravets

crteensy, 2016-09-15:
OS: Ubuntu 14.04 LTS
will try as soon as I can

commit 0351991e (2016-09-09): Update Teensy framework to 1.3.0; Add support for  Teensy 3.5/3.6; depend on new tool-teensy with teensy loader // Resolve #4 Resolve #5