platformio/platform-teensy #5: Teensyduino has been updated to 1.30

profra, 2016-09-08:
..... Teensy 3.5/3.6 included (beta):

https://github.com/PaulStoffregen/cores/tree/1.30

Please and thanks!

ivankravets, 2016-09-09:
What is your OS?

Please test development version:
```
platformio platform uninstall teensy
platformio platform install https://github.com/platformio/platform-teensy.git
```

Does it work with new Teensy?

closed 2016-09-09 by ivankravets

profra, 2016-09-10:
@ivankravets I am backer of new Teensy 3.5/3.6 on Kickstarter... Delivery of boards is expected only in October :-( ...I am working on two OS - Linux Mint(CLI) and W10(Atom)... I will let you know as soon as I have the boards.

blackketter, 2016-09-10:
I can confirm that a simple blink application compiles and uploads just fine to a Teensy 3.6 beta board with the development version.

ivankravets, 2016-09-10:
@blackketter  thanks for the report! What is your OS?

blackketter, 2016-09-10:
MacOS X 10.11.6

commit 0351991e (2016-09-09): Update Teensy framework to 1.3.0; Add support for  Teensy 3.5/3.6; depend on new tool-teensy with teensy loader // Resolve #4 Resolve #5