platformio/platform-teensy #6: Make Serial Monitor work with USB Type: "MIDI"

venelinpetrov, 2016-09-11:
Refer to this discussion: https://community.platformio.org/t/teensy-use-serial-monitor-when-usb-type-midi/742

blackketter, 2016-09-11:
Did you try adding -DUSB_MIDI_SERIAL to your platformio.ini build_flags line?

venelinpetrov, 2016-09-11:
@blackketter thanks for the answer. Unfortunately my project doesn't build if I change the build flag to `-DUSB_MIDI_SERIAL `, saying that `usbMIDI` is not defined in the current scope. Does this work on your side?

blackketter, 2016-09-11:
I was able to build with this platformio.ini and test serial midi sketch.  I'm not set up here to test the actual midi, but it builds and the serial works...

[serialmidi.zip](https://github.com/platformio/platform-teensy/files/466203/serialmidi.zip)

venelinpetrov, 2016-09-13:
I don't use the MIDI.h dependency. As I stated in my previous post I use `usbMIDI`. It seems to be automatically included with Arduino.h or something, which is the only header file in my project... Any insights?

jaard, 2017-08-08:
@venelinpetrov I'm getting the same error as you. Any progress in the meantime? Did you get it to work?

venelinpetrov, 2017-08-11:
@jaard I use `build_flags = -D USB_MIDI` and `usbMIDI`. With this configuration everything works nicely except the monitoring. I haven't tried any more blackketter's solution, because  my configuration is technically correct. I don't think I should change it. Better solution would be either to contribute and implement the functionality in question :) or back someone who has time and knowledge:
https://www.bountysource.com/teams/platformio/issues :)

ivankravets, 2018-09-10:
`teensy_serialmon` could be useful here, see https://github.com/platformio/platform-teensy/issues/36

venelinpetrov, 2018-09-10:
Nice! It's been a long time and it will be hard to test it now, but feel free to close the issue, because it's not directly related to PlatfotmIO anyways (and Paul said it will work ;) )