platformio/platform-teensy #7: teensy36 should use teensy_loader_cli and not Teensy.app for uploading

blackketter, 2016-09-19:
teensy31 uploads using the teensy_loader_cli binary, but teensy36 uses Teensy.app on MacOS
For consistency both should use teensy_loader_cli, like Linux.

MacOS: Sierra & El Capitan
PlatformIO: 3.0.1
framework-arduinoteensy: 1.130

I'd love to help fix this but I'm at a loss to see where this is specified.

ivankravets, 2016-09-27:
I opened an issue for that 18 days ago https://github.com/PaulStoffregen/teensy_loader_cli/issues/28

It looks that @PaulStoffregen doesn't have time for Teensy CLI

PaulStoffregen, 2016-09-28:
While working on Kickstarter, I don't have time for anything else.

Ping me next week....

ivankravets, 2016-09-28:
Thanks, @PaulStoffregen! We will wait!

ivankravets, 2016-10-08:
@PaulStoffregen could you take a look at Teensy CLI? Thanks!

ivankravets, 2016-11-08:
@PaulStoffregen do you have any news?

PaulStoffregen, 2016-11-09:
I've updated teensy_loader_cli.  You can now use it for all Teensy boards, on Windows, Linux and Mac.

Windows and Mac binaries are now available on the main page.

http://www.pjrc.com/teensy/loader_cli.html

commit e6e5a02b (2016-11-14): Update Teensy CLI package for 3.5 & 3.6 boards // Resolve #7
closed 2016-11-14 by ivankravets