platformio/platform-teensy #9: Problem with upload to Teensy 3.5

bule0017, 2016-12-18:
Hi,
I just bought Teensy 3.5 and I want to tesl Blink code.
My problem is that console give me this error:
`Uploading .pioenvs\teensy35\firmware.hex
Opening Teensy Loader...
Could not start Teensy Loader application! (r=1)
attempted: C:\Users\xxx\teensy.exe
*** [upload] Error 1
 [ERROR] Took 2.76 seconds`

But Teensy loader is installed in 
C:\Users\xxx xxx\.platformio\packages\tool-teensy

ivankravets, 2016-12-18:
Please provide a full output of `pio run -t upload -v` command. Thanks.

/cc @PaulStoffregen

bule0017, 2016-12-18:
Here it is:

`PS C:\Users\Goran Udovicic\Dropbox\PlatformIO\Teensy3.5\test> pio run -t upload -v
[12/18/16 21:40:36] Processing teensy35 (platform: teensy, board: teensy35, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------
Skip library with broken manifest: C:\Users\Goran Udovicic\.platformio\packages\framework-arduinoteensy\libraries\PS2Keyboard
Collected 86 compatible libraries
Looking for dependencies...
Project does not have dependencies
arm-none-eabi-g++ -o .pioenvs\teensy35\firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 -Wl,--defsym=__rtc_localtime=148209
3637 -fsingle-precision-constant --specs=nano.specs -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wl,-T"mk64fx512.ld" .pioenvs\teensy35\src\main.o "-LC
:\Users\Goran Udovicic\.platformio\packages\framework-arduinoteensy\cores\teensy3" -L.pioenvs\teensy35 -Wl,--start-group .pioenvs\teensy35\li
bFrameworkArduino.a -lm -lc -lgcc -larm_cortexM4l_math -Wl,--end-group

Checking program size
text       data     bss     dec     hex filename
16908       172    2696   19776    4d40 .pioenvs\teensy35\firmware.elf
arm-none-eabi-objcopy -O ihex -R .eeprom .pioenvs\teensy35\firmware.elf .pioenvs\teensy35\firmware.hex
teensy_post_compile -file=firmware -path="C:\Users\Goran Udovicic\Dropbox\PlatformIO\Teensy3.5\test\.pioenvs\teensy35" "-tools="C:\Users\Gora
n Udovicic\.platformio\packages\tool-teensy""
Opening Teensy Loader...
Could not start Teensy Loader application! (r=1)
attempted: C:\Users\Goran\teensy.exe
*** [upload] Error 1
======================================================== [ERROR] Took 2.54 seconds ========================================================
PS C:\Users\Goran Udovicic\Dropbox\PlatformIO\Teensy3.5\test>`

bule0017, 2016-12-18:
I just realize that upload works when I manually start teensy.exe loader

ivankravets, 2016-12-18:
The problem is linked with a space in your user account. I'll take a look where problem is hidden.

ivankravets, 2016-12-18:
Please go to `C:\Users\xxx xxx.platformio\platforms\teensy\builder\main.py` and replace [these lines](https://github.com/platformio/platform-teensy/blob/develop/builder/main.py#L196:L199) with

```
        UPLOADERFLAGS=[
            "-file=firmware", "-path=$BUILD_DIR",
            "-tools=%s" % (platform.get_package_dir("tool-teensy") or "")
        ],
```

bule0017, 2016-12-18:
Now open Teensy.exe loader but ther is another error:
`PS C:\Users\Goran Udovicic\Dropbox\PlatformIO\Teensy3.5\test> pio run -t upload -v
[12/18/16 21:57:44] Processing teensy35 (platform: teensy, board: teensy35, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------
Skip library with broken manifest: C:\Users\Goran Udovicic\.platformio\packages\framework-arduinoteensy\libraries\PS2Keyboard
Collected 86 compatible libraries
Looking for dependencies...
Project does not have dependencies
arm-none-eabi-g++ -o .pioenvs\teensy35\firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 -Wl,--defsym=__rtc_localtime=148209
4665 -fsingle-precision-constant --specs=nano.specs -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wl,-T"mk64fx512.ld" .pioenvs\teensy35\src\main.o "-LC
:\Users\Goran Udovicic\.platformio\packages\framework-arduinoteensy\cores\teensy3" -L.pioenvs\teensy35 -Wl,--start-group .pioenvs\teensy35\li
bFrameworkArduino.a -lm -lc -lgcc -larm_cortexM4l_math -Wl,--end-group

Checking program size
text       data     bss     dec     hex filename
16908       172    2696   19776    4d40 .pioenvs\teensy35\firmware.elf
arm-none-eabi-objcopy -O ihex -R .eeprom .pioenvs\teensy35\firmware.elf .pioenvs\teensy35\firmware.hex
teensy_post_compile -file=firmware -path=C:\Users\Goran Udovicic\Dropbox\PlatformIO\Teensy3.5\test\.pioenvs\teensy35 "-tools=C:\Users\Goran U
dovicic\.platformio\packages\tool-teensy"
TOeensy Loader could not find the file firmwarepening Teensy Loader...

*** [upload] Error 1
======================================================== [ERROR] Took 2.63 seconds ========================================================
PS C:\Users\Goran Udovicic\Dropbox\PlatformIO\Teensy3.5\test>`

ivankravets, 2016-12-18:
Please replace with this file 
* https://github.com/platformio/platform-teensy/blob/develop/builder/main.py

I've just tested in a Virtual machine and it works. I used "Sponge Bob" account.

closed 2016-12-18 by ivankravets

PaulStoffregen, 2016-12-18:
Please also make sure you have the latest version of Teensy Loader and related tools.

Version 1.32 had a bug which caused uploading problems with Arduino.  Whether it affects PlatformIO is a good question, but it's been fixed in 1.33.

ivankravets, 2016-12-18:
We use `teensy_loader_cli` for all host OS. Nevertheless, I see another issue where for Windows CLI is not used. Will fix it later.

@PaulStoffregen Are these binaries updated? https://www.pjrc.com/teensy/loader_cli.html

PaulStoffregen, 2016-12-18:
> Are these binaries updated?

They are version 2.1 from Nov 9th.  The 3 commits added after Nov 9 are not in those binaries.

ivankravets, 2016-12-18:
@PaulStoffregen are these 3 commits critical? Please ping me when you will recompile these binaries.

PaulStoffregen, 2016-12-18:
No, not critical.

bule0017, 2016-12-18:
Now it open Teensy.exe loader but error is: Teensy Loader could not find the file firmware
And Teensy.exe is now updated to 1.33 manually and in the bottom of loader window say firmware.hex (unreadable)

PaulStoffregen, 2016-12-18:
Teensy Loader has a hidden feature.  It's in Help > Verbose Info.  ...it's hidden in the Help menu... where nobody ever clicks..... ;)

The Verbose Info can tell you details, like what filename and path commands are being given.  Maybe if you look at that info, something will be obvious?  Perhaps the filename or path info isn't correct?

ivankravets, 2016-12-18:
@bule0017 Please update [this file](https://github.com/platformio/platform-teensy/blob/develop/builder/main.py) again. Now, PIO will use Teensy Loader CLI instead GUI.

ivankravets, 2016-12-18:
@PaulStoffregen I have a curious question... How is difficult to implement soft reset for Teensy Loader CLI?  PIO is cross-platform tool and could be used in various scenarios where there is no GUI or Teensy board is flashed remotely. 

We launched [PIO Remote](http://docs.platformio.org/en/latest/plus/pio-remote.html) 1 week ago. I see that developers can't use PIO Remote + Teensy because they can't reset board physically :(

Thanks!

bule0017, 2016-12-18:
Thank you very much. It's working now.
Only thing is `Soft reboot is not implemented for Win32`.. But that's not up to you.
So again thank you. I will try to donate to your team as soon as posible

ivankravets, 2016-12-18:
I wrote about `Soft reboot is not implemented for Win32` above. This is a critical issue for PlatformIO. You can't flash Teensy board without GUI (Linux, Raspberry Pi, etc) or remotely.  Let's wait for the answer from @PaulStoffregen.

PaulStoffregen, 2016-12-18:
> How is difficult to implement soft reset for Teensy Loader CLI?

Just run the teensy_reboot program, which you can find in Arduino's hardware/tools folder after installing Teensyduino.

ivankravets, 2016-12-18:
> Just run the teensy_reboot program, which you can find in Arduino's hardware/tools folder after installing Teensyduino.

Magic :) Will try it! It will be a big gift 🎄  for all Teensy developers 😊

ivankravets, 2016-12-19:
@PaulStoffregen 

```
 Uploading .pioenvs/teensy31/firmware.hex
Teensy Loader, Command Line, Version 2.1
Read ".pioenvs/teensy31/firmware.hex": 11528 bytes, 4.4% usage
Soft reboot is not implemented for OSX
Waiting for Teensy device...
(hint: press the reset button)
Found HalfKay Bootloader
Read ".pioenvs/teensy31/firmware.hex": 11528 bytes, 4.4% usage
Programming............
Booting
Rebooting...
Unable find Teensy Loader.  Is the Teensy Loader application running?
*** [upload] Error 1
```

Could you explain in details what we need to run? I mean how to resolve this 
> Soft reboot is not implemented for OSX
> Waiting for Teensy device...
> (hint: press the reset button)

PaulStoffregen, 2016-12-19:
Just run the teensy_reboot program, which you can find in Arduino's hardware/tools folder after installing Teensyduino.

PaulStoffregen, 2016-12-19:
Run teensy_reboot before teensy_loader_cli.

ivankravets, 2016-12-19:
@PaulStoffregen 

1. I have all files from Teensyduino. See our [tool-teensy](https://bintray.com/platformio/dl-packages/tool-teensy/1.3.0#files) packages
```
/Users/ikravets/.platformio/packages/tool-teensy
├── mktinyfat
├── package.json
├── teensy.app
│   └── Contents
│       ├── Info.plist
│       ├── MacOS
│       │   └── teensy
│       ├── PkgInfo
│       └── Resources
│           └── teensy.icns
├── teensy_gateway
├── teensy_loader_cli
├── teensy_post_compile
├── teensy_reboot
└── teensy_restart
```
2. We use teensy_loader_cli
3. See output if I call reboot before
```
...
Building .pioenvs/teensy31/firmware.hex
Rebooting...
Unable find Teensy Loader.  Is the Teensy Loader application running?
*** [upload] Error 1
```

PaulStoffregen, 2016-12-19:
Oh, opps, indeed teensy_reboot needs to communicate with the GUI-based Teensy Loader.

I'll add an option to run it stand-alone in a future version.  Sorry, I don't have anything more for you until then.

ivankravets, 2016-12-19:
Thanks, please don't forget to ping me about new changes. Thanks!

PaulStoffregen, 2016-12-19:
It will be in the next beta.  Use "teensy_reboot -s" to run stand-alone, without any communication to Teensy Loader.  You can also add "-v" for verbose output.

New betas are always announced here.
https://forum.pjrc.com/forums/6-Announcements

Sorry, I don't have a reliable way to directly ping you.  But you can click "Forum Tools" > "Subscribe To This Forum..." to get notifications when new announcements are posted.

PaulStoffregen, 2017-01-18:
Oh, yeah, this was added some time ago.  Use the new 1.35 release.  Run "teensy_reboot -s" for stand-alone mode.

commit a0290ee9 (2016-12-18): Fix uploading when user name contains spaces // Resolve #9