platformio/platform-teensy #10: Update framework-arduinoteensy

blackketter, 2017-01-12:
Teensyduino has been updated to 1.34:

https://www.pjrc.com/teensy/td_download.html

The most recent tag is see from @PaulStoffregen is 1.31:

https://github.com/PaulStoffregen/cores/tree/1.31

Thanks!

PaulStoffregen, 2017-01-12:
yup, looks like I'm not diligent about tagging releases....

blackketter, 2017-01-18:
@PaulStoffregen Congrats on 1.35.  Now would be a great time to tag that release.  👍

PaulStoffregen, 2017-01-18:
https://github.com/PaulStoffregen/cores/releases/tag/1.35

ivankravets, 2017-01-18:
@PaulStoffregen How about for soft reset for CLI tool?

ivankravets, 2017-01-18:
Fixed in https://github.com/platformio/platform-teensy/releases/tag/v1.6.0

closed 2017-01-18 by ivankravets

PaulStoffregen, 2017-01-18:
> How about for soft reset for CLI tool?

Like this?
https://github.com/PaulStoffregen/teensy_loader_cli/blob/master/teensy_loader_cli.c#L43

Hopefully that can help, because the reality is I'm not planning to do anything more with the command line loader in the foreseeable future.  So many other things are higher priority, like migrating Teensy to a newer version of gcc, and support for the 2nd USB port on Teensy 3.6, and much needed improvements in my long-neglected documentation.

ivankravets, 2017-01-18:
@PaulStoffregen I mean this https://github.com/platformio/platform-teensy/issues/9#issuecomment-267958924