platformio/platform-teensy #13: Use teensy_reboot script

ivankravets, 2017-01-27:
See @PaulStoffregen 

> Oh, yeah, this was added some time ago. Use the new 1.35 release. Run "teensy_reboot -s" for stand-alone mode.

https://github.com/platformio/platform-teensy/issues/9#issuecomment-273632303

mrkaroshi, 2017-03-13:
I think I'm missing something obvious - when I upload to the Teensy 3.1, after copying the most recent 1.35 Teensy applications into .platformio, I still get a message saying "Soft reboot is not implemented in OSX". Is there a workaround with instructions somewhere for this?

arneboon, 2017-03-16:
Hi @mrkaroshi,

Had problems too with uploading in PlatformIO using teensy_loader_cli. Here is the thread on the teensy [forum](https://forum.pjrc.com/threads/36546-teensy_loader_cli-on-OSX-quot-Error-opening-HID-Manager-quot?p=136909#post136909).

Now using the Teensy Loader implementation instead. This is how:

PlatformIO checks the ~/.platformio/packages/tool-teensy folder for existence of teensy_loader_cli. By deleting/renaming this file it skips to the Teensy Loader implementation. 

See source below in ~/.platformio/platforms/teensy/builder/main.py

Hope this helps. Would be nice if there is a more sexy way of selecting between the teensy_loader_cli or Teensy Loader. Perhaps there is, haven't found it yet.

```
if any([
        isfile(
            join(
                platform.get_package_dir("tool-teensy") or "",
                "teensy_loader_cli%s" % b)) for b in ("", ".exe")
]):
    env.Append(
        UPLOADER="teensy_loader_cli",
        UPLOADERFLAGS=[
            "-mmcu=$BOARD_MCU",
            "-w",  # wait for device to apear
            "-s",  # soft reboot if device not online
            "-v"  # verbose output
        ],
        UPLOADHEXCMD='$UPLOADER $UPLOADERFLAGS $SOURCES')
else:
    env.Append(
        REBOOTER="teensy_reboot",
        UPLOADER="teensy_post_compile",
        UPLOADERFLAGS=[
            "-file=firmware", '-path=$BUILD_DIR',
            '-tools=%s' % (platform.get_package_dir("tool-teensy") or "")
        ],
        UPLOADHEXCMD='$UPLOADER $UPLOADERFLAGS')
```

mrkaroshi, 2017-03-17:
@arneboon perfect! I suppose it's not exactly majestic, and I'm guessing I'll have to delete the CLI bin every time I update PlatformIO, but that's saved me a lot of hassle - I've been sat here with my project in pieces because I had to be able to reach the reset button. Thanks a million!

venelinpetrov, 2017-03-20:
All issues are related to Teensy 3.x, will this work for 2.0 too?

chappy1978, 2017-04-12:
I found that when you replace 
"teensy_loader_cli%s" % b)) for b in ("", ".exe") 
with
"Teensy%s" % b)) for b in ("", ".app")
It starts Teensy.app but has some errors and fails to work. Any thoughts

chappy1978, 2017-04-12:
I brought this problem up with Paul the maker of the Teensy. It doesnt look like the CLI will be MacOS rebootable ever. 
https://github.com/PaulStoffregen/teensy_loader_cli/issues/37

WasabiFan, 2017-12-20:
I can see that the flags mentioned above are passed to the teensy utilities [here](https://github.com/platformio/platform-teensy/blob/develop/builder/main.py#L210) yet I am still prompted to press the button on each deployment; what's missing to make the end-to-end scenario work? Is it something I could figure out myself and open a PR for?

I'm on Windows 10.

commit 42fe6288 (2018-02-07): Implement "teensy-gui" and "teensy-cli" upload protocols, allow switching between them // Resolve #13
closed 2018-02-07 by ivankravets