platformio/platform-teensy #14: float to string using snprintf and stringstreams

joe31093, 2017-01-28:
When using snprintf to convert a float to a string, the call fails.  Please see code and output below.
Hardware = Teensy3.6

```
#include <arduino.h>
int main()
{
	// Rates: 300, 600, 1200, 2400, 4800, 9600, 14400, 19200, 28800, 38400, 57600, or 115200
	Serial.begin(9600);

	int value1 = -12345;
	int value2 = 9876;
	float pi = 3.141596;
	char buffer[32];
	int copied = 0;

	while (true)
	{
		memset(buffer, 0x00, sizeof(buffer));
		copied = snprintf(buffer, sizeof(buffer), "%d", value1);
		Serial.print("copied: ");
		Serial.println(copied);
		Serial.print("buffer: ");
		Serial.println(buffer);

		memset(buffer, 0x00, sizeof(buffer));
		copied = snprintf(buffer, sizeof(buffer), "%d", value2);
		Serial.print("copied: ");
		Serial.println(copied);
		Serial.print("buffer: ");
		Serial.println(buffer);

		memset(buffer, 0x00, sizeof(buffer));
		copied = snprintf(buffer, sizeof(buffer), "%.4f", pi);
		Serial.print("copied: ");
		Serial.println(copied);
		Serial.print("buffer: ");
		Serial.println(buffer);

		delay(5000);
	}

	return 0;
}
```

Output:
copied: 6
```
buffer: -12345
copied: 4
buffer: 9876
copied: 0
buffer:
```

I tried using a template ostringstream method as well. It it crashes on ss << Number when the number is a float.

Code:
```
template std::string to_string(const T& Number)
{
std::ostringstream ss;
ss << Number;
return ss.str();
}
```

ivankravets, 2017-08-22:
Could you try https://github.com/platformio/platform-teensy/releases/tag/v2.4.0 ?

`pio update`

joe31093, 2017-08-24:
Just tried 2.4.0 and the code still exceptions.  The line asm(".global _printf_float"); is still needed to make this work.

commit 5100d81b (2018-02-07): Fix issue with floating point for printf/scanf // Resolve #14
closed 2018-02-07 by ivankravets