platformio/platform-teensy #16: SPI doesn't work on Teensy LC (works from Teensyduino)

alexozer, 2017-02-13:
Hi,

I've recently been testing Teensy LC with some SPI devices and have found that apparently SPI communication does not succeed when code is built and uploaded with platformio, yet it does seem to work fine when identical code is built and uploaded using Teensyduino. This behavior is only observable on Teensy LC; SPI code works fine on Teensy 3.1.

I've been able to narrow the issue to platformio using a [Bluefruit LE NRF8001](https://www.adafruit.com/product/1697) module from Adafruit, although I have also confirmed Teensy LC + platformio + SPI doesn't work with other SPI devices like the [RFM69](https://www.sparkfun.com/products/12775). For the Bluefruit I've used this sample code from Adafruit's tutorial to test:

```c++
// This version uses the internal data queing so you can treat it like Serial (kinda)!

#include <SPI.h>
#include "Adafruit_BLE_UART.h"

// Connect CLK/MISO/MOSI to hardware SPI
// e.g. On UNO & compatible: CLK = 13, MISO = 12, MOSI = 11
#define ADAFRUITBLE_REQ 10
#define ADAFRUITBLE_RDY 2     // This should be an interrupt pin, on Uno thats #2 or #3
#define ADAFRUITBLE_RST 9

Adafruit_BLE_UART BTLEserial = Adafruit_BLE_UART(ADAFRUITBLE_REQ, ADAFRUITBLE_RDY, ADAFRUITBLE_RST);
/**************************************************************************/
/*!
    Configure the Arduino and start advertising with the radio
*/
/**************************************************************************/
void setup(void)
{ 
  while (!Serial);
  Serial.begin(9600);
  Serial.println(F("Adafruit Bluefruit Low Energy nRF8001 Print echo demo"));

  BTLEserial.begin();
}

/**************************************************************************/
/*!
    Constantly checks for new events on the nRF8001
*/
/**************************************************************************/
aci_evt_opcode_t laststatus = ACI_EVT_DISCONNECTED;

void loop()
{
  // Tell the nRF8001 to do whatever it should be working on.
  BTLEserial.pollACI();
  
  // Ask what is our current status
  aci_evt_opcode_t status = BTLEserial.getState();
  // If the status changed....
  if (status != laststatus) {
    // print it out!
    if (status == ACI_EVT_DEVICE_STARTED) {
        Serial.println(F("* Advertising started"));
    }
    if (status == ACI_EVT_CONNECTED) {
        Serial.println(F("* Connected!"));
    }
    if (status == ACI_EVT_DISCONNECTED) {
        Serial.println(F("* Disconnected or advertising timed out"));
    }
    // OK set the last status change to this one
    laststatus = status;
  }
  
  if (status == ACI_EVT_CONNECTED) {
    // Lets see if there's any data for us!
    if (BTLEserial.available()) {
      Serial.print("* "); Serial.print(BTLEserial.available()); Serial.println(F(" bytes available from BTLE"));
    }
    // OK while we still have something to read, get a character and print it out
    while (BTLEserial.available()) {
      char c = BTLEserial.read();
      Serial.print(c);
    }
    
    // Next up, see if we have any data to get from the Serial console

    if (Serial.available()) {
      // Read a line from Serial
      Serial.setTimeout(100); // 100 millisecond timeout
      String s = Serial.readString();

      // We need to convert the line to bytes, no more than 20 at this time
      uint8_t sendbuffer[20];
      s.getBytes(sendbuffer, 20);
      char sendbuffersize = min(20, s.length());
      
      Serial.print(F("\n* Sending -> \"")); Serial.print((char *)sendbuffer); Serial.println("\"");
      
      // write the data
      BTLEserial.write(sendbuffer, sendbuffersize);
    }
  }
}
```

Uploading this with platformio yields over serial:

```
Adafruit Bluefruit Low Energy nRF8001 Print echo demo
Error in ACI Setup
```

Uploading this with Teensyduino yields:

```
Adafruit Bluefruit Low Energy nRF8001 Print echo demo
* Advertising started
* Connected!
```

ivankravets, 2017-08-22:
Could you try the latest https://github.com/platformio/platform-teensy/releases/tag/v2.4.0 ?

`pio update`

ivankravets, 2018-02-07:
Please reopen if you still need help

closed 2018-02-07 by ivankravets

alexozer, 2018-02-08:
@ivankravets I haven't worked on this for a while so I can't debug it at this time, and I haven't tried the latest release. Maybe that works.