platformio/platform-teensy #18: Error building (as: unrecognized option `-mcpu=cortex-m4')

amittel, 2017-03-06:
Just got this error. Couldn't find anything to fix it.
```
[03/06/17 11:39:11] Processing teensy31 (platform: teensy, board: teensy31, framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Collected 96 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <OctoWS2811> v1.3

Compiling .pioenvs\teensy31\src\Buzzer.o
Compiling .pioenvs\teensy31\src\Crosstimer.o
Compiling .pioenvs\teensy31\src\Digits.o
Compiling .pioenvs\teensy31\FrameworkArduino\AudioStream.o
Compiling .pioenvs\teensy31\FrameworkArduino\DMAChannel.o
Compiling .pioenvs\teensy31\FrameworkArduino\HardwareSerial1.o
Compiling .pioenvs\teensy31\FrameworkArduino\HardwareSerial2.o
Compiling .pioenvs\teensy31\FrameworkArduino\HardwareSerial3.o
as: unrecognized option `-mcpu=cortex-m4'
as: unrecognized option `-mcpu=cortex-m4'
*** [.pioenvs\teensy31\FrameworkArduino\DMAChannel.o] Error 1
*** [.pioenvs\teensy31\FrameworkArduino\AudioStream.o] Error 1
as: unrecognized option `-mcpu=cortex-m4'
*** [.pioenvs\teensy31\src\Buzzer.o] Error 1
as: unrecognized option `-mcpu=cortex-m4'
*** [.pioenvs\teensy31\FrameworkArduino\HardwareSerial1.o] Error 1
as: unrecognized option `-mcpu=cortex-m4'
*** [.pioenvs\teensy31\src\Digits.o] Error 1
as: unrecognized option `-mcpu=cortex-m4'
as: unrecognized option `-mcpu=cortex-m4'
*** [.pioenvs\teensy31\FrameworkArduino\HardwareSerial2.o] Error 1
*** [.pioenvs\teensy31\FrameworkArduino\HardwareSerial3.o] Error 1
as: unrecognized option `-mcpu=cortex-m4'
*** [.pioenvs\teensy31\src\Crosstimer.o] Error 1
 [ERROR] Took 2.15 seconds
```

Platform.ini
```
[env:teensy31]
platform = teensy
framework = arduino
board = teensy31
```

```
> pio --version
PlatformIO, version 3.2.1
```

Atom: IDE 1.7.3 | Core 3.2.1

Tried to update project, but didn't help.
Updated all Libs and Core, didn't help.

amittel, 2017-03-14:
Still getting the error.
I tried to comment out the line
https://github.com/platformio/platform-teensy/blob/develop/builder/main.py#L125

But then I get another error:
`as: unrecognized option -meabi=5`
But I can't find any `-meabi` appereance in _main.py_

ivankravets, 2017-08-22:
Please update to the latest 2.5.0 of dev platform (`pio update`).

How can we reproduce this issue?

amittel, 2017-11-09:
Just made an update and now it compiles.

`teensy @ 2.6.0`
`PlatformIO, version 3.4.1`

closed 2017-11-09 by amittel