platformio/platform-teensy #20: Confusing versioning

sstaub, 2017-04-03:
The Pio version (1.7.0) is very confusing because the actual Teensyduino version is in the moment 1.36.
The versions should generally follows the original versions, also on Arduino etcetera.

sstaub, 2017-04-10:
It seems that framework-arduinoteensy reflect the teensyduino version. If this right you can close this issue and please update to version 1.36

slugh23, 2017-04-20:
I'll second that.  The new Teensyduino v1.36 was released on 31 March.  Would it be possible please to update it from v1.35 to v1.36.

ivankravets, 2017-04-20:
@sstaub Teensy development platform consists of multiple frameworks including Arduino/Teensyduino. 

1.7.0 means a version of development platform, where 1.36 is a version of Arduino/Teensyduino framework.

commit 12965425 (2017-05-03): Update Arduino framework to v136 // Resolve #20
closed 2017-05-03 by ivankravets