platformio/platform-teensy #23: TeensyLoaderCLI won't upload anymore.

mgcrea, 2017-05-22:
Not sure what changed on my machine (up-to-date OSX on Sierra) but uploading sketches with `pio run -t upload -e teensy30 -v` via CLI does not work anymore, looks like the script does not see the teensy:

- Did update/upgrade platformio.
- Started a new project from scratch.
- Did reinstall both latest Arduino.app and Teensyduino

```
$ pio run -t upload -e teensy30 -v
[Mon May 22 18:28:08 2017] Processing teensy30 (platform: teensy, board: teensy30, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 90 compatible libraries
Looking for dependencies...
Project does not have dependencies
arm-none-eabi-g++ -o .pioenvs/teensy30/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 -Wl,--defsym=__rtc_localtime=1495470488 -fsingle-precision-constant --specs=nano.specs -Wl,-T"mk20dx128.ld" .pioenvs/teensy30/src/main.o -L/Users/olivier/.platformio/packages/framework-arduinoteensy/cores/teensy3 -L.pioenvs/teensy30 -Wl,--start-group .pioenvs/teensy30/libFrameworkArduino.a -lm -lc -lgcc -larm_cortexM4l_math -Wl,--end-group
Checking program size
text	   data	    bss	    dec	    hex	filename
11040	    168	   2032	  13240	   33b8	.pioenvs/teensy30/firmware.elf
arm-none-eabi-objcopy -O ihex -R .eeprom .pioenvs/teensy30/firmware.elf .pioenvs/teensy30/firmware.hex
teensy_loader_cli -mmcu=mk20dx128 -w -s -v .pioenvs/teensy30/firmware.hex
Teensy Loader, Command Line, Version 2.1
Read ".pioenvs/teensy30/firmware.hex": 11208 bytes, 8.6% usage
Error opening HID Manager
Soft reboot is not implemented for OSX
Waiting for Teensy device...
(hint: press the reset button
```
Uploading via GUI (auto-opened with Arduino.app) is working fine.

Thanks!

JMStudiosJoe, 2017-06-07:
is there a fix for this..?

ErinCall, 2017-07-08:
@mgcrea I'm having this problem as well. I had Karabiner installed; I uninstalled it (with `sudo /Library/Application\ Support/org.pqrs/Karabiner/uninstall.sh`) and rebooted, but I'm still getting "Unable to claim interface." Do you know if Karabiner's uninstaller might have left some config lines around?

billputer, 2017-08-10:
I was having the exact same issue with the QMK firmware and my Ergodox EZ, and uninstalling Karabiner Elements fixed it.

ivankravets, 2018-02-07:
Could you re-test with the latest development version https://github.com/platformio/platform-teensy#development-version ?

Does it work now via Teensy App?

ivankravets, 2018-02-08:
Does it work with the latest 3.0.0 version? Please run `pio update` and try again.

ivankravets, 2019-11-12:
https://github.com/platformio/platform-teensy/issues/54

closed 2019-11-12 by ivankravets