platformio/platform-teensy #24: Teensy Build Errors with std::deque

pierre-rouanet, 2017-05-29:
The following program does not compile on Teensy 3.1:

```Arduino
#include <Arduino.h>

#include <deque>

std::deque<int> v = {1, 2, 3};

void setup() {
  v.push_back(4);
}

void loop() {
}
```

With the build logs:
```bash
pio run -v

[Mon May 29 12:04:05 2017] Processing bugreport (platform: teensy; src_filter: +<examples/modules/test.cpp>; board: teensy31; framework: arduino)
-------------------------------------------------------------------------------------------------------------------
Collected 97 compatible libraries
Looking for dependencies...
Project does not have dependencies
arm-none-eabi-g++ -o .pioenvs/bugreport/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 -Wl,--defsym=__rtc_localtime=1496052245 -fsingle-precision-constant --specs=nano.specs -Wl,-T"mk20dx256.ld" .pioenvs/bugreport/src/examples/modules/test.o -L/Users/pierrerouanet/.platformio/packages/framework-arduinoteensy/cores/teensy3 -L.pioenvs/bugreport -Wl,--start-group .pioenvs/bugreport/libFrameworkArduino.a -lm -lc -lgcc -larm_cortexM4l_math -Wl,--end-group
/Users/pierrerouanet/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/libc_s.a(lib_a-signalr.o): In function `_kill_r':
signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
/Users/pierrerouanet/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/libc_s.a(lib_a-signalr.o): In function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
collect2: error: ld returned 1 exit status
*** [.pioenvs/bugreport/firmware.elf] Error 1
============================================ [ERROR] Took 1.20 seconds ============================================
```

It does compile using vector instead of deque (contrary to what seems to be indicated here: https://community.platformio.org/t/teensy-build-errors-with-std-vector/850)

ivankravets, 2017-08-22:
Does it compile with Teensyduino?

ivankravets, 2017-11-10:
See https://forum.pjrc.com/threads/29177-Teensy-3-1-signalr-c-(-text-_kill_r-0xe)-undefined-reference-to-_kill-error

```
#include <Arduino.h>

#include <deque>

extern "C"{
  int _getpid(){ return -1;}
  int _kill(int pid, int sig){ return -1; }
  int _write(){return -1;}
}

std::deque<int> v = {1, 2, 3};

void setup() {
  v.push_back(4);
}

void loop() {
}
```

closed 2017-11-10 by ivankravets