platformio/platform-teensy #30: Support for aarch64

ivankravets, 2018-01-09:
Related thread https://community.platformio.org/t/compiling-for-teensy-on-arm-fails/2863/

PaulStoffregen, 2018-05-31:
Support for ARM 64 bit is in the works too.  I have a Jetson TX2 here on my desk... ;)

ivankravets, 2018-05-31:
Please ping me here when toolchain will be ready. We will deploy it you our CDN storage

PaulStoffregen, 2018-08-04:
I worked on this in June, haven't touched it since then.

Here is the toolchain Teensyduino uses, with working build scripts for aarch64.
https://github.com/PaulStoffregen/ARM_Toolchain_2016q3_Source

I also updated the older toolchain source for the version Arduino uses for their SAM & SAMD boards.
https://github.com/PaulStoffregen/ARM_Toolchain_2014q1_Source

A beta test of Teensyduino+Arduino using this toolchain and everything else ported to aarch64 can be found here:
https://forum.pjrc.com/threads/52649-Teensyduino-1-42-Beta-8

I haven't touched the Jetson board or aarch64 stuff since mid-June.  In July I rewrote the Ethernet library and published version 2.0.0 (for all boards, not just Teensy).  I'm probably going to focus on hardware stuff in August & September.  Very unlikely I'll do anything more with aarch64 until Arduino actually makes a non-beta release.

If anyone is still watching this thread, at least you can know the toolchain and other stuff for Teensy was all ported several weeks ago and there is an experimental beta.

ivankravets, 2018-08-06:
@PaulStoffregen thank you so much for your efforts! I would like to create these missed packages but I can't. I don't have `aarch64` hardware and can't unpack https://www.pjrc.com/teensy/td_142-beta8/TeensyduinoInstall.linuxaarch64.xz

@olskar could you help us here? Could you download https://www.pjrc.com/teensy/td_142-beta8/TeensyduinoInstall.linuxaarch64.xz and provide toolchain and tools as `tar.gz`?

Thanks!

PaulStoffregen, 2018-08-06:
Here you go.  :)
https://github.com/PaulStoffregen/ARM_Toolchain_2016q3_Source/commit/94b37f9ad90a10c05d0b64757a6b0c7620a626d1

ivankravets, 2018-08-07:
@PaulStoffregen thanks! ;) Toolchain has been deployed to our CDN. @olskar please re-test.

The last step! We will need binary of https://github.com/PaulStoffregen/teensy_loader_cli . Could someone provide it?

bornae, 2019-05-25:
Did this issue ever get resolved? 

I am running into the following error trying to build for a teensy 3.5 on a Jetson TX2 with Ubuntu 16.04

`Compiling .pioenvs/teensy35/FrameworkArduino/usb_seremu.c.o
Compiling .pioenvs/teensy35/FrameworkArduino/usb_serial.c.o
Compiling .pioenvs/teensy35/FrameworkArduino/usb_touch.c.o
Compiling .pioenvs/teensy35/FrameworkArduino/yield.cpp.o
Archiving .pioenvs/teensy35/libFrameworkArduino.a
Indexing .pioenvs/teensy35/libFrameworkArduino.a
Linking .pioenvs/teensy35/firmware.elf
/home/nvidia/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: cannot find -larm_cortexM4lf_math
collect2: error: ld returned 1 exit status
*** [.pioenvs/teensy35/firmware.elf] Error 1
========================== [ERROR] Took 9.07 seconds ==========================
`

I'd appreciate any insight on how to resolve this issue. PlatformIO has been working great in this setup running on windows/Mac/desktop ubuntu.

Thank you,

bornae, 2019-05-25:
I have been able to download the latest version of arduin/teensyduino and build the project successfully that way.

bornae, 2019-05-25:
After some investigation it appears like some files are missing from the lib folder of the mbed toolchain.  
Copying these over to the Jetson does fix the issue (At least it build have not tried running yet)

however I am not familiar enough with the interworking of PlatformIO to create a pull request in a reasonable amount of time. 

Hope this helps. 

```
PWD.  :~/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/lib 
$ scp libarm_cortexM* nvidia@192.168.90.10:.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/lib/
nvidia@192.168.90.10's password: 
libarm_cortexM0l_math.a                                                                         100% 2207KB  25.1MB/s   00:00    
libarm_cortexM4l_math.a                                                                         100% 2694KB  27.7MB/s   00:00    
libarm_cortexM4lf_math.a                                                                        100% 2640KB  37.4MB/s   00:00    
(base) 
```

ivankravets, 2019-05-25:
@bornae we used a toolchain which @PaulStoffregen shared with us, see https://github.com/PaulStoffregen/ARM_Toolchain_2016q3_Source/commit/94b37f9ad90a10c05d0b64757a6b0c7620a626d1

Could share these `libarm*.a` files somewhere? We will repack toolchain. Thanks!

PaulStoffregen, 2019-05-25:
Oh, opps... yeah, the Teensyduino installer adds those files to the toolchain during install.

They're exactly the same files on all platforms, so perfectly fine to just copy the ones you have for Windows, Mac or any other Linux.

ivankravets, 2019-05-25:
Thanks! @bornae please re-test with `$ pio update`.

bornae, 2019-05-28:
Thank you @ivankravets and @PaulStoffregen 

I deleted my entire ~/.platformio folder to get rid of my modification and then tried a build again.
It successfully built and deployed. 

I appreciate the quick turnaround time.

ivankravets, 2019-05-28:
How did you flash a board? As I remember there is no teensy CLI for aarch64.

PaulStoffregen, 2019-05-28:
The teensy_loader_cli should compile and work on any Linux system, ARM or X86, 32 or 64 bits.

We now also publish the normal Teensyduino installer for aarch64.  It has the GUI version of Teensy Loader and everything else.

Also, unrelated but something you might wish to know: Ubuntu appears to have recently pushed an update to 18.04 which changes the filter policy for ModemManager, causing it to again interfere with Teensy and other boards.  I've updated to udev rules file we publish for Teensy.

https://www.pjrc.com/teensy/49-teensy.rules

If you have any copies of "49-teensy.rules", please update them.  If you get Linux users reporting partial loss of incoming serial data, and being unable to open the serial device some of the time, it's probably ModemManager causing trouble again.

ivankravets, 2019-05-28:
@bornae  could you compile `teensy_loader_cli` on your aarch64 machine and share binary somewhere? https://github.com/PaulStoffregen/teensy_loader_cli

> We now also publish the normal Teensyduino installer for aarch64

@PaulStoffregen could you upload somewhere the binary files? I don't have aarch64 machine. 

Thanks!

bornae, 2019-05-28:
teensy_loader_cli seems to work fine without me having to do anything special.

I did have to install the udev rules per Paul's instructions. 

``` bash
nvidia@tegra-ubuntu:~/.platformio/packages/tool-teensy  $ ls -al
total 528
drwx------ 2 nvidia nvidia   4096 May 28 07:22 .
drwxrwxr-x 6 nvidia nvidia   4096 May 28 07:22 ..
-rwxr-x--- 1 nvidia nvidia    263 Jun 16  2018 package.json
-rwxr-xr-x 1 nvidia nvidia 521644 Apr 28  2017 teensy_loader_cli
-rwxr-xr-x 1 nvidia nvidia     13 Jun 16  2018 teensy_reboot
nvidia@tegra-ubuntu:~/.platformio/packages/tool-teensy  $ cat package.json 
{
    "description": "Teensy Loader",
    "name": "tool-teensy",
    "system": [
        "linux_armv6l",
        "linux_armv7l",
        "linux_armv8l",
        "linux_aarch64"
    ],
    "url": "https://www.pjrc.com/teensy/loader.html",
    "version": "1.21.1"
}nvidia@tegra-ubuntu:~/.platformio/packages/tool-teensy  $ 

```

ivankravets, 2019-05-28:
Great! So, I close this issue!

closed 2019-05-28 by ivankravets