platformio/platform-teensy #31: Using extra_scripts custom PROGNAME is overwritten

ahpuck, 2018-01-25:
When I set a custom build file name using extra_scripts no matter what I set it to it gets overridden by the following lines in main.py
```
env.Replace(
...removing irrelevant lines
    PROGNAME="firmware",
    PROGSUFFIX=".elf"
)
```
then further down the same file
```
#
# Target: Build executable and linkable firmware
#

target_elf = None
if "nobuild" in COMMAND_LINE_TARGETS:
    target_firm = join("$BUILD_DIR", "firmware.hex")
else:
    target_elf = env.BuildProgram()
    target_firm = env.ElfToHex(join("$BUILD_DIR", "firmware"), target_elf)

AlwaysBuild(env.Alias("nobuild", target_firm))
target_buildprog = env.Alias("buildprog", target_firm, target_firm)
```

Is there any reason the PROGNAME is ignored? I'm happy to submit a PR with the necessary changes if there's no reason other than it hasn't been implemented yet.

Here's my platformio.ini
```
[platformio]
build_dir = bin

[env:teensy31]
; http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#projectconf-advanced-scripting
extra_scripts = pre:prebuildTasks.py
platform = teensy
board = teensy31
framework = arduino
lib_deps =
  EEPROM
  XBee
  LinkedList
  FastLED = 3.1.4
```

relevant section of prebuildTasks.py
```
gitVersion = subprocess.check_output("git tag --list | sort | tail -1", stderr=subprocess.STDOUT, shell=True).strip()
env.Replace(PROGNAME="firmware_%s-%s" % (gitVersion, nonReleaseVersion))
```
nonReleaseVersion is pulled from a JSON file.

ahpuck, 2018-01-25:
Ah I see that this has been fixed on develop!

ivankravets, 2018-01-25:
Yes, please use dev version. The final release is coming soon.

ahpuck, 2018-01-25:
Just a note - it truncates the PROGNAME in the file hex file. The output I get from a build is below. You can see it truncates firmware_1.2.5-143 to firmware_1.2.hex. It appears to happening somewhere on line 236. I believe in this command when I debug a little `target_firm = env.ElfToHex(target_elf)`. target_elf is correct. target_firm is truncated. 

```
Linking bin/teensy31/firmware_1.2.5-143
Checking program size
text       data     bss     dec     hex filename
119568        0   13368  132936   20748 bin/teensy31/firmware_1.2.5-143
Building bin/teensy31/firmware_1.2.hex
Uploading bin/teensy31/firmware_1.2.hex
Teensy Loader, Command Line, Version 2.1
Read "bin/teensy31/firmware_1.2.hex": 119568 bytes, 45.6% usage
Soft reboot is not implemented for OSX
Waiting for Teensy device...
(hint: press the reset button)
```

closed 2018-01-26 by ivankravets

ahpuck, 2018-01-26:
Awesome. Thanks Ivan!

commit f995e4f2 (2018-01-26): Allow to override PROGNAME // Resolve #31