platformio/platform-teensy #36: Support for `teensy_ports` tool

ivankravets, 2018-05-31:
See https://forum.pjrc.com/threads/52525-Teensyduino-1-42-Beta-6

```
teensy_ports -L
```

PaulStoffregen, 2018-05-31:
Teensyduino 1.42 also now comes with a teensy_serialmon program.  It is intended to implement "serial monitor" communication with Teensy.  You run it with "./teensy_serialmon {identifier}".  The communication is done on stdin and stdout, and it sends you status messages on stderr.

Of course you can communicate with Teensy as a serial device, **if** you have programmed it with the code configured to implement USB serial.

But teensy_serialmon adds many nice features.  If you configure the code to implement HID, Audio, MIDI or other modes without Serial, teensy_serialmon can automatically detect these.  It also knows how to detect USB device removal and connection.  If you have a Teensy programed as Audio, it will connect to the extra emulated serial interface.  If you then upload Serial code, it will automatically close the connection when the device disconnects (either a physical unplug or running teensy_reboot), and will be able to reconnect to it using the different protocol for whatever code you have just programmed.

ivankravets, 2018-05-31:
It seems that `teensy_serialmon` is a solution for https://github.com/platformio/platform-teensy/issues/6 ?

PaulStoffregen, 2018-05-31:
Yes, it solves the serial monitor issue in #6.

Of course you must compile the code with the correct -D args.  For Serial, you compile with -DUSB_SERIAL.  If you want to give users MIDI, you must change the compiler flags to use -DUSB_MIDI.

After programming, which Teensy begins running the freshly built code, it will turn Teensy's USB port back on.  Your PC does USB enumeration to detect the USB device.  Then teensy_serialmon can automatically detect which USB option you used and start giving you the communication on its stdin/stdout.

You can leave teensy_serialmon running while programming code.  Unlike normal serial usage, where you must close the port, teensy_serialmon does all that internally, even if you program new code with a different USB type!

mgcrea, 2018-06-14:
@PaulStoffregen not sure if this is the proper place to report but the `teens_ports` cli appears to be hanging on my mac:

```
$ /Applications/Arduino.app/Contents/Java/hardware/tools/teensy_ports -v
teensy_ports, verbose mode
not already running
socket created
bound to port 28542
USB device add callback
  loc=14130000, vid=16C0, pid=0483, ver=0200, ser=3872730
  actual serailnum=387273
    name: [no_device] (Teensy) Serial
USB device remove callback
Serial add callback
  name=/dev/cu.usbmodem3872730, loc=14130000, vid=16c0, pid=0483, ver=0200
  found prior teensy at this loc, age=0.002
    name: /dev/cu.usbmodem3872730 (Teensy) Serial
Serial remove callback
HID Manager started
```

Same thing for serialmon:

```
$ /Applications/Arduino.app/Contents/Java/hardware/tools/teensy_serialmon -v /dev/cu.usbmodem3872730
teensy_serialmon, loc = /dev/cu.usbmodem3872730
USB device add callback
  loc=14130000, vid=16C0, pid=0483, ver=0200, ser=3872730
  actual serailnum=387273
USB device remove callback
Serial add callback
  name=/dev/cu.usbmodem3872730, loc=14130000, vid=16c0, pid=0483, ver=0200
Serial remove callback
HID Manager started
```

Any idea?

PaulStoffregen, 2018-06-14:
What happens when you use Arduino 1.8.5 with Teensyduino 1.42?  Can you upload programs from Arduino?  If not, please use Teensy Loader's Help > Verbose Info window to capture the log of events.

mgcrea, 2018-06-14:
@PaulStoffregen everything is working fine on the Arduino side with 1.8.5&1.42 (upload via teensy-gui), for the `teensy_ports` I was missing the `-L` flag and it works.

Only thing is `teensy_serialmon -v /dev/cu.usbmodem3872730` that output nothing though my serial port has activity that I can properly see with miniterm, but I can see in the verbose logs that it correctly reconnects after upload/flash.

PaulStoffregen, 2018-06-14:
> Only thing is teensy_serialmon -v /dev/cu.usbmodem3872730 that output nothing

Run it like this:

teensy_serialmon 14130000

(or substitute whatever location string teensy_ports actually found)

mgcrea, 2018-06-15:
@PaulStoffregen understood, in my case though I don't have any output either:
> working with Miniterm:

```
λ ~/Projects/mgcrea/arduino-motor-driver @stepper *
$ make serial
--- Miniterm on /dev/cu.usbmodem3872500  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
! BUTTON: RISING_EDGE
> goMotor: HIGH
> goForwardMotor
! BUTTON: FALLING_EDGE
> stopMotor, position=-26602
```

> Not working with `teensy_serialmon` (no serial output):

```
λ ~/Projects/mgcrea/arduino-motor-driver @stepper *
$ /Applications/Arduino.app/Contents/Java/hardware/tools/teensy_serialmon -v 14130000
teensy_serialmon, loc = 14130000
USB device add callback
  loc=14130000, vid=16C0, pid=0483, ver=0200, ser=3872500
  actual serailnum=387250
USB device remove callback
Serial add callback
  name=/dev/cu.usbmodem3872500, loc=14130000, vid=16c0, pid=0483, ver=0200
Serial remove callback
HID Manager started
```

PaulStoffregen, 2018-06-15:
Try running "teensy_ports -L" (without -v).  What does it actually print?

Maybe you need "teensy_serialmon usb:14130000"?  The teensy_serialmon program is expecting the same string teensy_ports -L gives you.

PaulStoffregen, 2018-06-15:
The stings are different on all 3 platforms, and may change (especially on Macintosh, when/if Apple changes IOKit).  The only thing you can assume about the location string is no white space.

mgcrea, 2018-06-19:
@PaulStoffregen I can confirm that `teensy_serialmon usb:14130000` is properly working, thanks!