platformio/platform-teensy #38: Uploading to teensy gives error teensy_reboot not found (Raspberry Pi)

ivankravets, 2018-06-16:
Moved from https://community.platformio.org/t/uploading-to-teensy-gives-error-teensy-reboot-not-found-raspberry-pi/4384

------

@PaulStoffregen could you share somewhere `teensy_reboot` binary? I know that source code is closed for this tool. Thanks!

PaulStoffregen, 2018-06-16:
https://forum.pjrc.com/threads/52805-Uploading-to-teensy-with-platformio-gives-error-teensy_reboot-not-found-(Raspberry-Pi?p=181664&viewfull=1#post181664

cudmore, 2018-06-16:
Thanks @ivankravets, I upgraded platformio and it now can upload to my teensy.

Upgrade
```
sudo pip uninstall platformio
sudo pip install -U platformio
```
Run
```
sudo platformio run --target clean
sudo platformio run --target upload
```
p.s. Still don't understand why I/we need `sudo` for platformio?

ivankravets, 2018-06-16:
Update vs Upgrade