platformio/platform-teensy #41: Teensy 3.2 cannot be programmed after first time, unless press reset button

424778940z, 2018-12-14:
I'm using latest stable version platformio ide

This issue may caused by I set it to usb uart mode, and download script doesn't reboot it before program.

i did following change in this file: .atom\packages\platformio-ide\penv\Lib\site-packages\platformio\builder\scripts\teensy.py
start at: Line 68

before:

``` python
env.Append(
        UPLOADER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_post_compile"),
        UPLOADERFLAGS=[
            "-file=firmware",
            '-path="$BUILD_DIR"',
            '-tools="%s"' % join("$PIOPACKAGES_DIR", "tool-teensy")
        ],
        UPLOADHEXCMD='"$UPLOADER" $UPLOADERFLAGS'
    )
```

after:

``` python
env.Append(
        REBOTTER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_reboot"),
        UPLOADER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_post_compile"),
        UPLOADERFLAGS=[
            "-file=firmware",
            '-path="$BUILD_DIR"',
            '-tools="%s"' % join("$PIOPACKAGES_DIR", "tool-teensy")
        ],
        UPLOADHEXCMD='"$REBOTTER" & "$UPLOADER" $UPLOADERFLAGS'
    )
```

it solve this issue for now but will have a error because delay between restart and program is not enough
error msg: "Teensy Loader is currently busy with another operation (p).  Please try again in a few seconds, or restart Teensy Loader."

ivankravets, 2018-12-14:
@PaulStoffregen which minimum version of Teensy Loader is required for Teensy 3.5?  I see that the latest package in our repository is 1.41.

ivankravets, 2018-12-14:
@zmrubin the source code is here https://github.com/platformio/platform-teensy/blob/develop/builder/main.py#L197:L227

PaulStoffregen, 2018-12-14:
> Its strange, because my very first program uploaded no problem at all. Any help is appreciated!

On that first try, perhaps Teensy is still running a program that was written by Arduino?

When you switch to Arduino, is a button press needed for the first upload?  Then does Arduino upload automatically?  On that first Arduino upload the Teensy would still running be running code that was created by PlatformIO.

If you can confirm both of these cases, that would suggest PlatformIO is indeed sending the reboot request properly, but creating images on the Teensy which do not respond to the request.

scott620, 2018-12-14:
> On that first try, perhaps Teensy is still running a program that was written by Arduino?
>
> When you switch to Arduino, is a button press needed for the first upload? Then does Arduino upload [...]

OK.. i just uploaded a program to the board using PlatformIO (had to use the flash button). Right after this I opened Arduino and without even closing PlatformIO the Arduino IDE will auto-reboot and flash the Teensy with no issue, no button press necessary.

PaulStoffregen, 2018-12-14:
Ok then, there goes that theory.  Looks like PlatformIO is creating the correct code.

Any chance this is Windows 7?  (or XP or 8... anything before 10 there Microsoft finally fixed the surprise removal bugs)

PaulStoffregen, 2018-12-14:
If it's one of those older buggy versions of Windows, maybe whether PlatformIO still have a handle to the COM port open could be playing a factor?

scott620, 2018-12-14:
> Ok then, there goes that theory. Looks like PlatformIO is creating the correct code.
> 
> Any chance this is Windows 7? (or XP or 8... anything before 10 there Microsoft finally fixed the surprise removal bugs)

Actually Mac Mojave, is that a problem? Should I try it on my windows 10 machine and see if I can replicate the failure?

PaulStoffregen, 2018-12-14:
So far, the only problem I know on Mojave is a regression on ability to handle sustained maximum full speed incoming USB data.  Still unknown whether it's a Java issue or something MacOS.  To be realistic, it's on my to-do list for a few months from now.  Currently working on an urgent profit (the kind that actually funds my time to do open source the rest of the year) so can't do a lot right now.

But yeah, try Windows 10.  Avoid earlier pre-10 versions of Windows.  They all have USB bugs.

Also, does PlatformIO support using the HID emulated serial?  (like when you select MIDI or RawHID in Arduino Tools > USB Type)

ivankravets, 2018-12-14:
@PaulStoffregen could you take a look at what we actually do? See https://github.com/platformio/platform-teensy/blob/develop/builder/main.py#L214:L227

1. We call `teensy_post_compile`
2. We call `teensy_reboot` without any flags.

I see that in a case with CLI upload mode, we use `teensy_reboot -s`. Is it OK?

scott620, 2018-12-14:
> So far, the only problem I know on Mojave is a regression on ability to handle sustained maximum full speed incoming USB data. Still unknown whether it's a Java issue or something MacOS. To be [...]

Alright, I tested it on windows 10 and the same problems occur. I cant flash with platformio unless i press the flash button AND have the teensy flasher program up in front of the platformio window. When i open Arduino IDE and attempt to flash it works every time.

PaulStoffregen, 2018-12-14:
>     1. We call `teensy_post_compile`
>     2. We call `teensy_reboot` without any flags.

I can confirm that is the correct approach.

Sorry, can't do anything more at this moment.  Big hardware project, already far behind schedule....

However, one thing that might help is comparing the exact commands.  In Arduino, click File > Preferences and turn on verbose output while uploading.  That should at least make the commands Arduino is using visible.

424778940z, 2018-12-15:
just got email from this issue after 2 years lol.

I think the problem still same as before, the teensy is not being rebooted, you could try write a simple bliking led code in pio, then load it to teensy.

then keep try load it again, see if the led still blinking at the same time, if it does it means teensy it not being rebooted. then we could try copy the teense tools from android to pio folder see if their version works.

closed 2018-12-19 by ivankravets

ivankravets, 2018-12-19:
@PaulStoffregen actually, you were right. I see that you use just 1 `teensy_post_compile` now. I did the same changes and it works for me now for Teensy 3.1 and Teensy 3.6

@scott620 @424778940z @zmrubin please re-test with the latest upstream version https://docs.platformio.org/en/latest/platforms/teensy.html#upstream

Also, please don't forget to update all packages via `pio update` or from PIO Home.

ivankravets, 2018-12-20:
Please update to the final 3.7.0 release https://community.platformio.org/t/teensy-dev-platform-v3-7-0/5975

scott620, 2018-12-21:
> @PaulStoffregen actually, you were right. I see that you use just 1 `teensy_post_compile` now. I did the same changes and it works for me now for Teensy 3.1 and Teensy 3.6
>
> @scott620 @424778940z [...]

Works perfectly now!

424778940z, 2018-12-21:
sadly I don't have a treensy anymore, my teensy was losted with a drone project.... I cannot test it this time

commit cab465bd (2018-12-19): Switch to built-in rebooter in "teensy_post_compile" script // Resolve #41