platformio/platform-teensy #42: Board id changed for teensy 2.0

Rotzbua, 2018-12-29:
Affected version: 3.7.0
Working version: 3.6.0

>  Error: Invalid value for "-b" / "--board": teensy20. Please search for board ID using platformio boards command

Commit https://github.com/platformio/platform-teensy/commit/c46cfe2b0396f17d95677cf105d5327e4dac4e36 breaks backwards compatibility. Now I get an error and *every* build script has to be changed to new value.

I suggest a revert to old `teensy20` which make more sense because teensy 3.0 is similar named as `teensy30`. 

Otherwise teensy 3.0 has to renamed from `teensy30` to `teensy3`...

ivankravets, 2019-01-30:
We did that to be compatible with Teensyduino boards.txt. Its upload tool depends on board ID and needs `teensy2`. This is actually our bug which we fixed. We don't plan to change this board ID later. Please update your CI configuration to use new board ID.

Thanks!

closed 2019-01-30 by ivankravets

Rotzbua, 2019-01-30:
Did the simplest solution, removed teensy.