platformio/platform-teensy #43: Teensy LC builds much larger using PlatformIO than Arduino IDE

e3b0c442, 2019-01-31:
Hello,

I am experiencing an issue where builds for Teensy LC in PlatformIO are much larger than builds done in the Arduino IDE itself.

Some examples:

An empty sketch (default starting point for both Arduino and PlatformIO):
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Arduino:
```
Sketch uses 6044 bytes (9%) of program storage space. Maximum is 63488 bytes.
Global variables use 2088 bytes (25%) of dynamic memory, leaving 6104 bytes for local variables. Maximum is 8192 bytes.
```

PlatformIO:
```
DATA:    [==        ]  23.7% (used 1940 bytes from 8192 bytes)
PROGRAM: [=         ]  11.7% (used 7420 bytes from 63488 bytes)
```

Bare program which loops and outputs "hello world" on Serial:
```c++
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("hello world");
  delay(1000);
}
```

Arduino:
```
Sketch uses 6964 bytes (10%) of program storage space. Maximum is 63488 bytes.
Global variables use 2104 bytes (25%) of dynamic memory, leaving 6088 bytes for local variables. Maximum is 8192 bytes.
```

PlatformIO:
```
DATA:    [====      ]  37.3% (used 3056 bytes from 8192 bytes)
PROGRAM: [==        ]  18.3% (used 11596 bytes from 63488 bytes)
```

Program using the Teensy ADC library which reads and then prints a value:
```c++
#include <Arduino.h>
#include <ADC.h>

ADC *adc = new ADC();
void setup()
{
  // put your setup code here, to run once:
  adc->setResolution(10);
  adc->setAveraging(16);
  adc->setConversionSpeed(ADC_CONVERSION_SPEED::VERY_LOW_SPEED);
  adc->setSamplingSpeed(ADC_SAMPLING_SPEED::VERY_LOW_SPEED);
  pinMode(A0, INPUT);

  Serial.begin(115200);
}

void loop()
{
  // put your main code here, to run repeatedly:
  int val = adc->analogRead(A0);
  Serial.printf("The ADC read %d\n", val);
  delay(1000);
}
```

Arduino:
```
Sketch uses 13636 bytes (21%) of program storage space. Maximum is 63488 bytes.
Global variables use 2224 bytes (27%) of dynamic memory, leaving 5968 bytes for local variables. Maximum is 8192 bytes.
```

PlatformIO:
```
DATA:    [====      ]  41.8% (used 3424 bytes from 8192 bytes)
PROGRAM: [======    ]  60.8% (used 38608 bytes from 63488 bytes)
```

I am using the "Smallest Code" preset in the Arduino IDE, which per the board definition sets the following flags:
```
teensyLC.build.flags.common=-g -Wall -ffunction-sections -fdata-sections -nostdlib
teensyLC.build.flags.dep=-MMD
teensyLC.build.flags.cpu=-mthumb -mcpu=cortex-m0plus -fsingle-precision-constant
teensyLC.build.flags.defs=-D__MKL26Z64__ -DTEENSYDUINO=146
teensyLC.build.flags.cpp=-fno-exceptions -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti
teensyLC.build.flags.c=
teensyLC.build.flags.S=-x assembler-with-cpp
teensyLC.build.flags.ld=-Wl,--gc-sections,--relax,--defsym=__rtc_localtime={extra.time.local} "-T{build.core.path}/mkl26z64.ld" -lstdc++
teensyLC.build.flags.libs=-larm_cortexM0l_math -lm
teensyLC.menu.opt.osstd.build.flags.optimize=-Os --specs=nano.specs
teensyLC.menu.opt.osstd.build.flags.ldspecs=
```

Looking at the build output from PlatformIO, it seems like most if not all of these flags are being set (I'm still trying to compare build outputs). The one I noticed missing was `--specs=nano.specs`; I tried setting that in the build_flags environment setting to no avail.

e3b0c442, 2019-01-31:
OK, after digging into the docs again I think Advanced Scripting will fit my need. I'll leave it to the team whether adding this option by default wants to be explored or not.

Thanks.

ivankravets, 2019-01-31:
@PaulStoffregen what is default build mode for Teensyduino? Does it use `nano.specs` by default?

PaulStoffregen, 2019-01-31:
It's changed many times.  I'm not even sure right now, without looking...

Easiest way to look with Arduino IDE, using "show verbose info" while compiling, and then click Verify.

ivankravets, 2019-01-31:
[image]

# Boards.txt

The default mode is "Faster", which does not use `--specs=nano.specs`.

```
teensy36.menu.opt.o2std=Faster
teensy36.menu.opt.o2std.build.flags.optimize=-O2
teensy36.menu.opt.o2std.build.flags.ldspecs=
teensy36.menu.opt.o2lto=Faster with LTO
teensy36.menu.opt.o2lto.build.flags.optimize=-O2 -flto -fno-fat-lto-objects
teensy36.menu.opt.o2lto.build.flags.ldspecs=-fuse-linker-plugin
teensy36.menu.opt.o1std=Fast
teensy36.menu.opt.o1std.build.flags.optimize=-O1
teensy36.menu.opt.o1std.build.flags.ldspecs=
teensy36.menu.opt.o1lto=Fast with LTO
teensy36.menu.opt.o1lto.build.flags.optimize=-O1 -flto -fno-fat-lto-objects
teensy36.menu.opt.o1lto.build.flags.ldspecs=-fuse-linker-plugin
teensy36.menu.opt.o3std=Fastest
teensy36.menu.opt.o3std.build.flags.optimize=-O3
teensy36.menu.opt.o3std.build.flags.ldspecs=
teensy36.menu.opt.o3purestd=Fastest + pure-code
teensy36.menu.opt.o3purestd.build.flags.optimize=-O3 -mpure-code -D__PURE_CODE__
teensy36.menu.opt.o3purestd.build.flags.ldspecs=
teensy36.menu.opt.o3lto=Fastest with LTO
teensy36.menu.opt.o3lto.build.flags.optimize=-O3 -flto -fno-fat-lto-objects
teensy36.menu.opt.o3lto.build.flags.ldspecs=-fuse-linker-plugin
teensy36.menu.opt.o3purelto=Fastest + pure-code with LTO
teensy36.menu.opt.o3purelto.build.flags.optimize=-O3 -mpure-code -D__PURE_CODE__ -flto -fno-fat-lto-objects
teensy36.menu.opt.o3purelto.build.flags.ldspecs=-fuse-linker-plugin
teensy36.menu.opt.ogstd=Debug
teensy36.menu.opt.ogstd.build.flags.optimize=-Og
teensy36.menu.opt.ogstd.build.flags.ldspecs=
teensy36.menu.opt.oglto=Debug with LTO
teensy36.menu.opt.oglto.build.flags.optimize=-Og -flto -fno-fat-lto-objects
teensy36.menu.opt.oglto.build.flags.ldspecs=-fuse-linker-plugin
teensy36.menu.opt.osstd=Smallest Code
teensy36.menu.opt.osstd.build.flags.optimize=-Os --specs=nano.specs
teensy36.menu.opt.osstd.build.flags.ldspecs=
teensy36.menu.opt.oslto=Smallest Code with LTO
teensy36.menu.opt.oslto.build.flags.optimize=-Os -flto -fno-fat-lto-objects --specs=nano.specs
teensy36.menu.opt.oslto.build.flags.ldspecs=-fuse-linker-plugin
```

ivankravets, 2019-01-31:
I've just release a new version 3.7.1 which allows to control firmware optimization => https://github.com/platformio/platform-teensy/releases/tag/v3.7.1

1. Please run `pio update`
2. See documentation https://docs.platformio.org/en/latest/platforms/teensy.html#optimization

closed 2019-01-31 by ivankravets

PaulStoffregen, 2019-01-31:
Those are the defaults for Teensy 3.6.

Teensy LC has similar settings, but the default is different (due to its small memory).

e3b0c442, 2019-01-31:
Tried this out, works great. Thanks for the fast turnaround 👍

commit 79372486 (2019-01-31): Allow to control firmware optimization // Resolve #43
commit 48c8dc7a (2019-01-31): Allow to control firmware optimization // Resolve #43