platformio/platform-teensy #44: --upload-port flag has no effect

yosemitebandit, 2019-03-30:
I have a machine connected to both a teensy 3.6 and a teensy 3.2 (both over USB):

```
% pio device list  
/dev/ttyACM1
------------
Hardware ID: USB VID:PID=16C0:0483 SER=4350050 LOCATION=1-1.3:1.0
Description: USB Serial

/dev/ttyACM0
------------
Hardware ID: USB VID:PID=16C0:0483 SER=4454990 LOCATION=1-1.2:1.0
Description: USB Serial

/dev/ttyAMA0
------------
Hardware ID: 3f201000.serial
Description: ttyAMA0
```

Initially I can also see both ttys:

```
% ls /dev/ttyACM*
/dev/ttyACM0  /dev/ttyACM1
```

When I try to upload to the 3.6, it appears as though the 3.2 is flashed.  (Prior to this, I was able to confirm that `ttyACM0` is the 3.6).

```
pio run --target upload --upload-port /dev/ttyACM0
```

This seems to brick the 3.2 (it has to be power-cycled and manually reflashed). The 3.2's tty is gone:

```
% ls /dev/ttyACM*
/dev/ttyACM0
```

Running the same upload command again will successfully upload to the 3.6 (presumably because that is the only available tty at this point?). 

I found this similar-sounding issue on the core repo, fwiw: https://github.com/platformio/platformio-core/issues/1904

yosemitebandit, 2019-03-30:
Also fwiw, the `--port` flag with the monitor command seems to work.  This will connect me to the 3.2 as expected:

`pio device monitor --echo --port /dev/ttyACM1`

pfeerick, 2019-11-02:
This sounds like the multi-board issue that was raised on the forum, which stems from the Teensy Loader not allowing a port to be specified, hence any port you specify via platformIO upload_port won't take. The OP for that thread was on linux, so instead turned the ports on and off programatically to get around the issue. An alternative might be to use [tycmd ](https://github.com/Koromix/tytools) via an extra_script to do the upload, as it seems it can be given specific port references or hardware IDs.

https://community.platformio.org/t/multiple-boards-upload-port-works-but-crashes-2nd-board/9029

oomek, 2020-05-14:
Is there any solution to this issue for Windows? I almost bricked my Teensy 2.0 by flashing it with Teensy 3.2 program. It's very annoying to be forced to constantly replugging cables when I'm working on 2 different boards simutaneously and it's very easy to make a mistake.

yosemitebandit, 2020-05-14:
@oomek  our team now uses PIO for builds and tycmd for uploads.

oomek, 2020-05-14:
Right, completely forgot about it. I used it as an easy consumer side updater. Is there any guide of how to integrate it with PIO?

pfeerick, 2020-05-14:
> @oomek our team now uses PIO for builds and tycmd for uploads.

Did you just do an extra_script that over-rides the `UPLOAD_CMD` or did you add in a [custom upload tool](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-upload-tool)?

oomek, 2020-05-14:
I did not change anything. I used tycmd before as stadalone with a batch file. In PIO I have just standard `upload_protocol = teensy-cli`

pfeerick, 2020-05-14:
Sorry, was editing my prior comment, and didn't realise it was tagging you instead of yosemitebandit. Anyway, see above for one possible way to 'integrate' tycmd ...

oomek, 2020-05-14:
Thank you. I'm gonna try it out tomorrow

collinedington, 2020-05-14:
@oomek I work with @yosemitebandit , here's the way we use tycmd:

We keep our boards in a fixed location on the linux machine (for us a raspberry pi).  This means the same board is always in the bottom right position, or the top left, etc.

First build tycmd [following the instructions in the github readme](https://github.com/Koromix/tytools).  I had trouble getting this to work on a mac but it was a breeze on Raspbian.

Then you can list your boards (using verbose flag) with `tycmd list -v` and you'll get something like this:

add 5530090-Teensy Teensy 3.6 (USB Serial)
  location: usb-1-1-2
  capabilities:
    unique
    run
    reboot
    serial

  interfaces:
    Serial: /dev/ttyACM0

That location information tells you where in the USB tree the board is.   I suspect this will work even through USB hubs but I haven't tried it yet.  There are [instructions on the tytools github](https://github.com/Koromix/tytools#using-tycmd) about how to use the various location tags as well, in this case we need `@`.  Then you can point tycmd at a *.hex or *.elf file and upload to that location like so:

`tycmd upload --board @usb-1-1-2 path/to/file.hex`

We still use platformio to compile the project, and you can locate the hex build in:
`project-dir/.pioenvs/teensy36/firmware.hex` [in PIO 3.6.7]
`project-dir/.pio/build/teensy36/firmware.hex` [in PIO 4.3.3]

tycmd is also nice in that it will check the teensy board version you're trying to upload to, and refuse to upload incompatible code, like something compiled for a 3.2 onto a 3.6 (PIO still doesn't do this and will happily KO a teensy if you send it the wrong code).

We haven't tried integrating like @pfeerick suggested above, but we'll give it a try!

oomek, 2020-05-14:
For some reason `upload_flags = --board "1610490-Teensy"` did not work
Flashed with ``upload_flags = -B"1610490-Teensy"`` switch succesfuly. Thanks!

pfeerick, 2020-05-15:
Yeah, finally got around to trying it myself...

`upload_flags = --board "1610490-Teensy"` doesn't work because it becomes `tycmd.exe upload "--board "6170180-Teensy"" .pio\build\teensy40\firmware.hex` and whilst tycmd correctly parses it, it seems the long form of options fail (like `Unknown option '--board 6170180-Teensy'`) if there is a `"` in front of them... 

If you add a space before the `--` of `--board`, it works. Or you can do `--board=1610490-Teensy` as that seems to work also if you want the long form.

This is with the extrascript mentioned above with the path to my executable...

upload_with_tycmd.py
```
Import("env")

env.Replace(
    UPLOADER="E:\\Tools\\tytools\\tycmd.exe",
    UPLOADCMD="$UPLOADER upload $UPLOADERFLAGS $SOURCE"
)
```

platformio.ini
```
[env:teensy40]
platform = teensy
board = teensy40
framework = arduino
extra_scripts = upload_with_tycmd.py
upload_protocol = custom
upload_flags = --board=6170180-Teensy
```

hmaarrfk, 2020-09-23:
I don't really suggest using the "port" since it tends to change. Instead I use the serial number as reported by the USB port.

I've made a PR to `teensy_loader_cli`, not sure if that is what platformio uses under the hood.

https://github.com/PaulStoffregen/teensy_loader_cli/pull/57 

If you use conda, and conda-forge, I've release it https://github.com/conda-forge/teensy_loader_cli-feedstock

But it seems like tycmd is more powerful. Maybe I'll try to build that for the conda ecosystem.

Unfortunately, I don't have a development machine for windows easily accessible, so I likely won't be making the necessary changes there.

dgorissen, 2022-08-15:
Just got bitten by this and the clear answer on https://askubuntu.com/questions/1374435/how-to-program-multiple-teensys-in-platformio-simultaneously-on-ubuntu worked for me, without the need for a python script