platformio/platform-teensy #46: Teensy35 Stream.timedRead() is protected

paulhayes, 2019-07-30:
This method is called by the following library: ArduinoHttpClient @ src\HttpClient.cpp:576:17

The method is public in the main arduino library and other compatables, should Stream not be symmetrical across the different platforms? Could you modify this to match arduino framework?

ArduinoHttpClient is listed as being compatible with Teensy, and it is except for this one methods protection level.

All the best

valeros, 2019-07-30:
Hi @paulhayes ! I'd better open the issue somewhere here https://github.com/PaulStoffregen/cores, as we are not responsible for developing and maintaining Arduino frameworks.

closed 2019-07-30 by valeros

paulhayes, 2019-08-21:
Thanks for pointing me in the right direction.