platformio/platform-teensy #47: Teensy 4.0 board support

wcalvert, 2019-08-05:
I would like to request that support for Teensy 4.0 be added. Teensy 4.0 has not been officially released yet, but it looks like the release is coming up very soon, after going through a very lengthy beta testing period. See this page for a comment from Paul that the release is soon: https://forum.pjrc.com/threads/54711-Teensy-4-0-First-Beta-Test?p=211461&viewfull=1#post211461

Paul's github has commits to the Arduino implementation: https://github.com/PaulStoffregen/cores/tree/master/teensy4

wcalvert, 2019-08-07:
Teensy 4.0 has been officially released: https://www.pjrc.com/store/teensy40.html

blackketter, 2019-08-14:
And a version of Teensyduino with support for 4.0 has been officially released:

https://www.pjrc.com/teensy/td_download.html

extesy, 2019-08-14:
Pinging @ivankravets and @valeros.

ivankravets, 2019-08-14:
We will add support soon. Sorry for the delay.

kisom, 2019-08-15:
I have a board definition done, but it requires updating https://bintray.com/platformio/dl-packages/framework-arduinoteensy/ to the latest version.

ivankravets, 2019-08-15:
@kisom do you have this board? Does it work if you update `framework-arduinoteensy` to the latest version?

kisom, 2019-08-15:
@ivankravets I do have the board:

```
$ pio platform update
...
Updating framework-arduinoteensy         @ 1.145.0        [Up-to-date]
```

Dropping the board JSON into my `~/.platformio` directory sees the board

```
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy 4.3.0 > Teensy 4.0
HARDWARE: IMXRT1062 396MHz, 512KB RAM, 1.94MB Flash
```

but fails with

```
ttributeError: 'SConsEnvironment' object has no attribute 'ElfToHex':
  File "/home/kyle/.local/lib/python2.7/site-packages/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/kyle/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/kyle/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/home/kyle/.platformio/platforms/teensy/builder/main.py", line 144:
    target_firm = env.ElfToHex(join("$BUILD_DIR", "${PROGNAME}"), target_elf)
```

kisom, 2019-08-15:
If I had to guess, I'd think the error might be due to the missing linker script in `framework-arduinoteensy`.

valeros, 2019-08-15:
I think `teensy4` board should be taken into account [here](https://github.com/platformio/platform-teensy/blob/develop/builder/main.py#L91)

kisom, 2019-08-15:
@valeros Thanks; I've updated it.

ivankravets, 2019-08-16:
1. Please open PlatformIO IDE Terminal and type 
```
pio update
```
We need updated toolchains.
2. Use development version http://docs.platformio.org/en/latest/platforms/teensy.html#upstream

Does it work?

kisom, 2019-08-16:
@ivankravets it doesn't:

https://imgur.com/a/mdEmucm

ivankravets, 2019-08-16:
@PaulStoffregen do you have any ideas? We use the latest core + toolchains.

ivankravets, 2019-08-16:
@valeros it seems like a problem in board manifest. See https://github.com/platformio/platform-teensy/blob/develop/boards/teensy40.json

Boards.txt
```
teensy40.name=Teensy 4.0
teensy40.upload.maximum_size=2031616
teensy40.upload.maximum_data_size=1048576
teensy40.upload.tool=teensyloader
teensy40.upload.protocol=halfkay
teensy40.build.board=TEENSY40
teensy40.build.core=teensy4
teensy40.build.mcu=imxrt1062
teensy40.build.warn_data_percentage=99
teensy40.build.toolchain=arm/bin/
teensy40.build.command.gcc=arm-none-eabi-gcc
teensy40.build.command.g++=arm-none-eabi-g++
teensy40.build.command.ar=arm-none-eabi-gcc-ar
teensy40.build.command.objcopy=arm-none-eabi-objcopy
teensy40.build.command.objdump=arm-none-eabi-objdump
teensy40.build.command.linker=arm-none-eabi-gcc
teensy40.build.command.size=arm-none-eabi-size
teensy40.build.flags.common=-g -Wall -ffunction-sections -fdata-sections -nostdlib
teensy40.build.flags.dep=-MMD
teensy40.build.flags.optimize=-Os
teensy40.build.flags.cpu=-mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16
teensy40.build.flags.defs=-D__IMXRT1062__ -DTEENSYDUINO=147
teensy40.build.flags.cpp=-std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing
teensy40.build.flags.c=
teensy40.build.flags.S=-x assembler-with-cpp
teensy40.build.flags.ld=-Wl,--gc-sections,--relax "-T{build.core.path}/imxrt1062.ld"
teensy40.build.flags.libs=-larm_cortexM7lfsp_math -lm -lstdc++
teensy40.build.fcpu=600000000
teensy40.serial.restart_cmd=false
teensy40.menu.usb.serial=Serial
teensy40.menu.usb.serial.build.usbtype=USB_SERIAL
#teensy40.menu.usb.keyboard=Keyboard
#teensy40.menu.usb.keyboard.build.usbtype=USB_KEYBOARDONLY
#teensy40.menu.usb.keyboard.fake_serial=teensy_gateway
#teensy40.menu.usb.touch=Keyboard + Touch Screen
#teensy40.menu.usb.touch.build.usbtype=USB_TOUCHSCREEN
#teensy40.menu.usb.touch.fake_serial=teensy_gateway
#teensy40.menu.usb.hidtouch=Keyboard + Mouse + Touch Screen
#teensy40.menu.usb.hidtouch.build.usbtype=USB_HID_TOUCHSCREEN
#teensy40.menu.usb.hidtouch.fake_serial=teensy_gateway
#teensy40.menu.usb.hid=Keyboard + Mouse + Joystick
#teensy40.menu.usb.hid.build.usbtype=USB_HID
#teensy40.menu.usb.hid.fake_serial=teensy_gateway
#teensy40.menu.usb.serialhid=Serial + Keyboard + Mouse + Joystick
#teensy40.menu.usb.serialhid.build.usbtype=USB_SERIAL_HID
#teensy40.menu.usb.midi=MIDI
#teensy40.menu.usb.midi.build.usbtype=USB_MIDI
#teensy40.menu.usb.midi.fake_serial=teensy_gateway
#teensy40.menu.usb.midi4=MIDIx4
#teensy40.menu.usb.midi4.build.usbtype=USB_MIDI4
#teensy40.menu.usb.midi4.fake_serial=teensy_gateway
#teensy40.menu.usb.midi16=MIDIx16
#teensy40.menu.usb.midi16.build.usbtype=USB_MIDI16
#teensy40.menu.usb.midi16.fake_serial=teensy_gateway
#teensy40.menu.usb.serialmidi=Serial + MIDI
#teensy40.menu.usb.serialmidi.build.usbtype=USB_MIDI_SERIAL
#teensy40.menu.usb.serialmidi4=Serial + MIDIx4
#teensy40.menu.usb.serialmidi4.build.usbtype=USB_MIDI4_SERIAL
#teensy40.menu.usb.serialmidi16=Serial + MIDIx16
#teensy40.menu.usb.serialmidi16.build.usbtype=USB_MIDI16_SERIAL
#teensy40.menu.usb.audio=Audio
#teensy40.menu.usb.audio.build.usbtype=USB_AUDIO
#teensy40.menu.usb.audio.fake_serial=teensy_gateway
#teensy40.menu.usb.serialmidiaudio=Serial + MIDI + Audio
#teensy40.menu.usb.serialmidiaudio.build.usbtype=USB_MIDI_AUDIO_SERIAL
#teensy40.menu.usb.serialmidi16audio=Serial + MIDIx16 + Audio
[... 29 lines cut ...]
teensy40.menu.opt.o3std.build.flags.optimize=-O3
teensy40.menu.opt.o3std.build.flags.ldspecs=
#teensy40.menu.opt.o3purestd=Fastest + pure-code
#teensy40.menu.opt.o3purestd.build.flags.optimize=-O3 -mpure-code -D__PURE_CODE__
#teensy40.menu.opt.o3purestd.build.flags.ldspecs=
#teensy40.menu.opt.o3lto=Fastest with LTO
#teensy40.menu.opt.o3lto.build.flags.optimize=-O3 -flto -fno-fat-lto-objects
#teensy40.menu.opt.o3lto.build.flags.ldspecs=-fuse-linker-plugin
#teensy40.menu.opt.o3purelto=Fastest + pure-code with LTO
#teensy40.menu.opt.o3purelto.build.flags.optimize=-O3 -mpure-code -D__PURE_CODE__ -flto -fno-fat-lto-objects
#teensy40.menu.opt.o3purelto.build.flags.ldspecs=-fuse-linker-plugin
teensy40.menu.opt.ogstd=Debug
teensy40.menu.opt.ogstd.build.flags.optimize=-Og
teensy40.menu.opt.ogstd.build.flags.ldspecs=
#teensy40.menu.opt.oglto=Debug with LTO
#teensy40.menu.opt.oglto.build.flags.optimize=-Og -flto -fno-fat-lto-objects
#teensy40.menu.opt.oglto.build.flags.ldspecs=-fuse-linker-plugin
teensy40.menu.opt.osstd=Smallest Code
teensy40.menu.opt.osstd.build.flags.optimize=-Os --specs=nano.specs
teensy40.menu.opt.osstd.build.flags.ldspecs=
#teensy40.menu.opt.oslto=Smallest Code with LTO
#teensy40.menu.opt.oslto.build.flags.optimize=-Os -flto -fno-fat-lto-objects --specs=nano.specs
#teensy40.menu.opt.oslto.build.flags.ldspecs=-fuse-linker-plugin

```

kisom, 2019-08-16:
@ivankravets Reading the error messages and the diff in 732b515d323d265f7983a30cbabdd85fe892c6f6, it looks like it's being compiled with the IMXRT1052, but it should be compiled for the IMXRT1062.

ivankravets, 2019-08-16:
Yes, @valeros will fix. Sorry.

kisom, 2019-08-16:
@ivankravets and @valeros thank you for the work you've put into this - it's much appreciated!

PaulStoffregen, 2019-08-17:
Definitely delete anything with "1052".  That's the chip we used in the very early beta.  Only 19 of those boards "exist in the wild".

While we're talking of Teensy 4.0, I'll briefly mention a potential known problem.  The linker file over-promises the RAM size:

    MEMORY
    {
        ITCM (rwx):  ORIGIN = 0x00000000, LENGTH = 512K
        DTCM (rwx):  ORIGIN = 0x20000000, LENGTH = 512K
        RAM (rwx):   ORIGIN = 0x20200000, LENGTH = 512K
        FLASH (rwx): ORIGIN = 0x60000000, LENGTH = 1984K
    }

The chip only has 1024K RAM.  The tricky part is the first 512K is dynamically assignable between ITCM and DTCM.  The startup code actually configures the RAM partition, based on how much was really used for ITCM.

But the GNU linker can't understand such a strange RAM partitioning arrangement.  Or if it can, I have no idea how to accomplish that.  So we're telling the linking a small lie, that the chip has more RAM.  If the user tries to actually consume more than 512K total in ITCM & DTCM, the code will crash.

In a future software version, I'm planning to add a utility which checks the amount of memory the linker actually used in both those sections.  It will give a build error if too much memory was used.  So far, no work has been done on this utility program.  I don't have a time frame right now.  I'm explaining this now, so you can be aware that a future version of Teensyduino will change the build process to add a post-linker check that ITCM+DTCM allocation is not too large.

valeros, 2019-08-17:
@kisom I've updated the board manifest, please retest with the latest upstream version.

blackketter, 2019-08-17:
Thanks.  I can confirm that I can build and upload blink!

wcalvert, 2019-08-17:
Compilation is looking good here, thank you for the quick response.

ivankravets, 2019-08-17:
The final is planned for Monday.

closed 2019-08-17 by ivankravets