platformio/platform-teensy #49: Teensy 4.0 compiler errors

loughmiller, 2019-08-20:
This happens when adding FastLED to any project.  It works under Teensy31, fails when I change the board to a Teesny40.

.platformio/packages/framework-arduinoteensy/libraries/DmxSimple/DmxSimple.cpp: In function 'void dmxBegin()':
.platformio/packages/framework-arduinoteensy/libraries/DmxSimple/DmxSimple.cpp:82:11: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   dmxPort = portOutputRegister(digitalPinToPort(dmxPin));
           ^
.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp: In constructor 'SoftwareSerial::SoftwareSerial(uint8_t, uint8_t, bool)':
.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp:75:11: error: 'Serial8' was not declared in this scope
   port = &Serial8;
           ^
.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp:95:8: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  txreg = portOutputRegister(digitalPinToPort(txPin));
        ^
.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial/SoftwareSerial.cpp:96:8: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  rxreg = portInputRegister(digitalPinToPort(rxPin));
        ^
*** [.pio/build/teensy40/lib7ea/DmxSimple/DmxSimple.cpp.o] Error 1
*** [.pio/build/teensy40/lib620/SoftwareSerial/SoftwareSerial.cpp.o] Error 1

loughmiller, 2019-08-21:
After digging in a bit more, it's clearly an issue with the arduinoteensy libraries or some definition inside platformio.  I can get it to compile under the Arduino IDE, bit not platformio.  I'm not sure exactly where to file it, but it clearly doesn't belong here, so I'll close the ticket.  Happy to file it in the correct place if you can point me there.

valeros, 2019-08-22:
Hi @loughmiller ! Could you please put together a minimal example to reproduce the issue?

loughmiller, 2019-08-22:
Just `#include <FastLED.h>` in any project and you'll get the compilation errors, but here's a short main.cpp that works and displays 10 red LEDs on the Teensy 3.2, but fails to compile when using the Teensy 4.0:

`#include <Arduino.h>
#include <FastLED.h>

#define NUM_LEDS 10
#define DISPLAY_LED_PIN 12
CRGB leds[NUM_LEDS];

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  FastLED.addLeds<NEOPIXEL, DISPLAY_LED_PIN>(leds, NUM_LEDS).setCorrection( 0xFFD08C );

  for (uint_fast16_t i = 0; i < NUM_LEDS; i++) {
    leds[i] = CHSV(0, 255, 100);
  }

  FastLED.show();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  FastLED.delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  FastLED.delay(500);
}`

loughmiller, 2019-08-22:
I'm running:

Platform.io - Home 2.2.0, Core 4.0.0
Teensy - 4.4.0
FastLED - 3.3.1

closed 2019-08-26 by ivankravets

ivankravets, 2019-08-27:
Thanks for the report! We have just fixed this issue.  The final release of this dev/platform is planed for tomorrow. You will need to switch [LDF](http://docs.platformio.org/en/latest/librarymanager/ldf.html) to `chain+` which handles C/C++ preprocessor constructions.

```
[env:teensy40]
platform = teensy
framework = arduino
board = teensy40
lib_ldf_mode = chain+
# build_flags = -D CORE_TEENSY
```

If you don't want to wait for the final release, just uncomment `build_flags = -D CORE_TEENSY` in `platformio.ini` above.

ivankravets, 2019-08-27:
I forgot, you need to update all PIO Core apckages via `pio update` command or PIO IDE > Quick Links > Updates > Updates ALL

commit 9fcf9814 (2019-08-26): Add "CORE_TEENSY" to SCons scope // Resolve #49