platformio/platform-teensy #50: Using Keyboard.press from a .cpp file

thekevinbrown, 2019-09-03:
Hi there!

I'm trying to use Teensy's keyboard emulation. I've put `build_flags = -DUSB_SERIAL_HID` into `platformio.ini`, but when I try to use `Keyboard.press()` from the main cpp file, I get:

```
error: 'Keyboard' does not name a type
```

The only examples I can find use `.ino` files, but I'm hoping to keep the `.cpp` extension for all files in the project. Is there an include or extern declaration I need?

valeros, 2019-09-03:
Hi @thekevinbrown ! Please attach here your code.

thekevinbrown, 2019-09-04:
Figured it out.

For anyone else stumbling onto this:
```c++
#include <usb_keyboard.h>
```
gives other compilation units access to the Keyboard instance.

closed 2019-09-04 by thekevinbrown