platformio/platform-teensy #51: Error compiling Arduino board "teensy40" on Raspbian GNU/Linux 10 (buster)

blackketter, 2019-09-27:
I'm trying to build a relatively simple  blink Arduino sketch on a Raspberry Pi 4 running Raspbian GNU/Linux 10 (buster) with PlatformIO, version 4.0.3.

The platformio.ini file for that environment is simple:

```
[env:teensy40]
platform = teensy
board = teensy40
framework = Arduino
```

First error is this:
```
Compiling .pio/build/teensy40/FrameworkArduino/Blink.cc.o
In file included from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:44:0,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from /tmp/tmpO5s8xX:1:
/home/dean/.platformio/packages/toolchain-gccarmnoneeabi@1.50401.0/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
```

Code is here: https://github.com/blackketter/blink.git but it happens with other projects.

No errors building on macOS.

blackketter, 2019-09-29:
Verified that build completes fine on Ubuntu 18.04.3 LTS 64-bit x86

blackketter, 2020-02-22:
Confirmed that this is still an issue with latest platformio packages.  Also, noted that teensy31 builds fine.

limitz, 2020-03-16:
I have the same problem when trying to compile on my nvidia jetson. I think it's the combination of required toolchain and host architecture. When it installs toolchain
```
 36     "toolchain-gccarmnoneeabi": {
 37       "type": "toolchain",
 38       "optional": false,
 39       "version": "~1.50401.0"
 40     },
```
it finds a package for your processor / architecture (in my case aarch64). But this package seems weird to me. It is a 5.4.1 toolchain that has the c++config header missing, whereas the earlier version 1.4 for aarch64 has version 5.4.3. Neither work, and newer packages for this toolchain don't exist. 

I don't feel comfortable just taking any c++config.h without finding out why it doesn't work to begin with.

jimsynz, 2020-04-28:
I'm seeing this also when compiling on armhf (in this case trying to compile in a docker container):

```
In file included from /root/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:45:0,
                 from /root/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /root/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from .pio/libdeps/teensy40/FreeRTOS-Teensy4_ID6737/src/FreeRTOS_TEENSY4.h:12,
                 from src/telemetry.hpp:4,
                 from src/logger.hpp:4,
                 from src/logger.cpp:1:
/root/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
*** [.pio/build/teensy40/src/logger.cpp.o] Error 1
In file included from /root/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:45:0,
                 from /root/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /root/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from .pio/libdeps/teensy40/FreeRTOS-Teensy4_ID6737/src/FreeRTOS_TEENSY4.h:12,
                 from src/main.cpp:1:
/root/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
*** [.pio/build/teensy40/src/main.cpp.o] Error 1
========================= [FAILED] Took 135.03 seconds =========================
```

Happy to provide any additional information or logs as required.

valeros, 2020-04-30:
Looks like the toolchains for ARM-based devices is outdated. Could someone install Teensyduino and upload somewhere bundled toolchains for the following systems:
* Linux Installer (ARM 32 bit / Raspberry Pi)
* Linux Installer (AARCH64 / Jetson TX2)

Thanks!

bornae, 2020-06-14:
I just ran into the same issue, 
```
/home/user/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
```
This is trying to compile for a teensy40 on a Jetson Xavier NX.
Would appreciate some help here.

valeros, 2020-06-17:
Hi @bornae ! I'd be great if you could install Teensyduino and upload somewhere bundled toolchain. Thanks!

maxgerhardt, 2020-08-24:
A workaround is available as described in https://community.platformio.org/t/teensy-4-0-not-compiling-on-raspberry-pi-bits-c-config-h-no-such-file-or-directory/15592. 

Basically, force usage of a newer `toolchain-gccarmnoneeabi` toolchain, copy the missing `libarm_cortexM7lfsp_math.a` file into the compiler folder and use my self-compiled and extracted `tool-teensy` package so that uploading is fixed, and you should get a working Raspberry Pi + Teensy 4 setup.

peterpolidoro, 2020-10-01:
Hi. It seems that this is still an issue. I am trying your workaround @maxgerhardt  (thanks for that!) but now it appears that the toolchain link on bintray is broken. The new link seems to be:
[https://bintray.com/platformio/tool-packages/download_file?file_path=3579c14-toolchain-gccarmnoneeabi-linux_armv6l-1.90301.200702.tar.gz](https://bintray.com/platformio/tool-packages/download_file?file_path=3579c14-toolchain-gccarmnoneeabi-linux_armv6l-1.90301.200702.tar.gz)
When I run it with the new link I get:

> toolchain-gccarmnoneeabi @ 1.90301.200702 has been installed!
> tool-teensy @ 1.152.200516 has been installed!
> sh: 1: arm-none-eabi-g++: not found
> *** Error 127

Spencer-Brennessel, 2020-10-05:
Hey! I'm in the same boat. I was the person who originally asked the question on PlatformIO and now my setup isn't working with a 404 error on that link. Please let me know if someone finds a new workaround or link. Thanks!

maxgerhardt, 2020-10-05:
@peterpolidoro Being unable to run `arm-none-abi-g++` is an indication of a binary incomaptibility. Are you running a 64-bit OS on your Rasbi? Then you need the AArch64 version (or install some 32-bit compatibility, don't know if that's there for ARMv6l though). The link for that is https://bintray.com/platformio/tool-packages/download_file?file_path=5966cc5-toolchain-gccarmnoneeabi-linux_aarch64-1.90301.200702.tar.gz

maxgerhardt, 2020-10-06:
For my and @Spencer-Brennessel Raspberry Pi this was repordedly fixed with the `platformio.ini`

```
[env:teensy40]
platform = teensy
framework = arduino
board = teensy40 
platform_packages = 
   toolchain-gccarmnoneeabi@https://bintray.com/platformio/tool-packages/download_file?file_path=3579c14-toolchain-gccarmnoneeabi-linux_armv6l-1.90301.200702.tar.gz
   tool-teensy@https://github.com/maxgerhardt/pio-tool-teensy-arm/archive/master.zip
```

and copying the missing `libarm_cortexM7lfsp_math.a` file from [here](https://community.platformio.org/t/teensy-4-0-not-compiling-on-raspberry-pi-bits-c-config-h-no-such-file-or-directory/15592/7?u=maxgerhardt) to the `.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/lib` folder (or `toolchain-gccarmnoneeabi@<hash>` folder in case the new compiler got downloaded there).

@valeros  Can you add my `tool-teensy` [package](https://github.com/maxgerhardt/pio-tool-teensy-arm) to the official pacakges with `platformio` as owner? This solves one compatibility issue.

peterpolidoro, 2020-10-12:
@maxgerhardt thank you, I was indeed trying to use the 32-bit version instead of the proper 64-bit version. So I switched to the 64-bit link and copied the missing `libarm_cortexM7lfsp_math.a` file into the proper directory. Now I am getting the error that it cannot find teensy_reboot.

Separately I am also having issues with platformio using outdated versions of some of the dependencies, despite having run `pio lib install` and `pio lib update` and watching all of the dependencies update properly.

maxgerhardt, 2020-10-12:
>Now I am getting the error that it cannot find teensy_reboot.

Can you share the output of `pio run -v -t upload` (to e.g. pastebin.com). 

Especially important is that it recognized the version of my tool-teensy package as 

```
PACKAGES:
 - framework-arduinoteensy 1.153.0 (1.53)
 - tool-teensy 1.152.200516 (1.52)
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
```
at the beginning.

maxgerhardt, 2020-10-12:
If the project compiles, then the `toolchain-gccarmnoneeabi` package works. The output about `tool-teensy` would be interesting.

Also, what exact Pi and OS (`uname -a`) are you running?

peterpolidoro, 2020-10-12:
I will collect more of the output for you and put it in another message. Just for reference, I am using a Raspberry Pi 4 B with 4GB RAM, running Ubuntu server 20.04 (5.4.0-1019-raspi #21-Ubuntu SMP PREEMPT Mon Sep 14 07:20:34 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux).

peterpolidoro, 2020-10-19:
Thank you I will try it out as soon as I have a little time. I would like to combine this with [tycmd](https://koromix.dev/tytools) so I can build and upload code to multiple teensy boards from a raspberry pi. Can the platformio.ini file links be to local files instead of bintray? I am thinking of putting the entire build environment into a Docker image to easily download and run. I would like to have all dependencies in the image so it will not need internet access after the initial Docker image download. Do you have to run the platformio.ini file once in order to install the toolchain, then copy over the missing file, then run it again? Is there a better way to install the toolchain and your tool-teensy? Thanks again for your great work on this.

maxgerhardt, 2020-10-23:
@peterpolidoro 

>I would like to combine this with tycmd so I can build and upload code to multiple teensy boards from a raspberry pi. Can the platformio.ini file links be to local files instead of bintray?

Yes, instead of writing `platform_packages = tool-teensy@http://...`, you can use the `file://` pseudo-protocol to just reference a path, e.g. if the repo is cloned at `/opt/fixed-tool-teensy`, write `file:///opt/fixed-tool-teensy`. Same goes for the `toolchain-gccarmnoneeabi` package. (PS: I'm not 100% sure whether a path to folder works directly *or* whether it needs a path to the `.tar.gz` installation package which is created by `pio package pack`. Try if in doubt.)

If you want to use `tycmd` for upload: You should compile the program for the Pi, put it in some path, then use [Advanced scripting](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-upload-tool) to override the `UPLOADCMD`  so that it instead of `teensy_cli` it executes a command of your choice, which should of course then be the appropiate command for `tycmd` to upload to a specific board. 

Then there's the question of how you determine to which board to upload, since the program needs a board name or serial number. You can e.g. solve this by having separate environments for each target board, then adding `build_flags` in  each environment in which the target serial number is contained. E.g. with `build_flags = -D TEENSY_UPLOAD_SERIAL=ABCDEF..`. You can then inspect `env["CPPDEFINES"]` (or in `projenv`) to retrieve that value in the upload script, see [example](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#construction-environments).

bornae, 2020-12-26:
> Hi @bornae ! I'd be great if you could install Teensyduino and upload somewhere bundled toolchain. Thanks!
@valeros Finally got some time to get everything back up and running. 

I am trying to build and upload to a teensy 4.0 from a Jetson Xavier NX.
Here is the error I am getting:
```
In file included from /home/jetson/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:45:0,
                 from /home/jetson/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /home/jetson/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from src/CorrectedTime.h:8,
                 from src/ CorrectedTime.cpp:5:
/home/jetson/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
```

Using the latest PlatformIO as of right now. 

```
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.11.1) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.153.0 (1.53) 
 - tool-teensy 1.21.1 
 - toolchain-gccarmnoneeabi 1.50401.1 (5.4.1)
 ```

I have attached the Arduino + Teensyduino install folder that I have verified working.

https://www.dropbox.com/t/MFDtP7fylKlgqUyq

maxgerhardt, 2020-12-26:
@bornae have you tried the `platformio.ini` suggestions from above? From your output it's visible that you're still using arm-none-eabi-gcc 5.4.1 which won't work. What's your current full `platformio.ini`?

bornae, 2020-12-26:
> @bornae have you tried the `platformio.ini` suggestions from above? From your output it's visible that you're still using arm-none-eabi-gcc 5.4.1 which won't work. What's your current full `platformio.ini`?

@maxgerhardt I appear that have missed that post. 

My initial plarformio.ini file here: 
```
[env:teensy40]
platform = teensy
framework = arduino
board = teensy40
targets = upload
upload_protocol = teensy-cli
```

I have added platform packages there and now I am missing the math library that you mentioned. I imagine if I get that in, it will end up working. 

However, between the time of my previous post and now, I was able to make pio work by patching the .platformio directory. 
I copied the toolchain from my teensyduino install into the .platformio folder, and also built the teensyloader-cli and replaced the included one. 

That combination is working even with the 5.4.1 arm-none-eabi-gcc 5.4.1

If it is helpful, I can upload the patched ~/.platformio folder that is working for me as well. 

Would like to know what the best path is to get a fix merged in here. I don't quite understand the inner-workings to put up a merge request.

valeros, 2021-01-06:
Hi @bornae! Could you please upload the toolchain from the Teensyduino somewhere? Thanks in advance.

bornae, 2021-01-06:
> Hi @bornae! Could you please upload the toolchain from the Teensyduino somewhere? Thanks in advance.

It is the dropbox link in my post  (About 5 posts back) 

https://www.dropbox.com/t/MFDtP7fylKlgqUyq
Here it is again. Thank you.

valeros, 2021-01-06:
Thanks, I uploaded `aarch64` packages to the CDN. Would be great if you could try to reinstall the teensy packages from scratch (just delete or rename `~/.platformio` folder) and try to compile your project again.

I suspect there we still have an outdated toolchain package for ARMV6, so it will be great if someone uploads the toolchain package from Teensyduino for ARMV6.

blackketter, 2021-01-06:
I tried deleting my .platformio directory on my Raspberry Pi 4 and rebuilding.  Failed with the following error:

```
/home/dean/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
```
Full output:

```
dean@pi4:~/src/blink $ pio run -e teensy40
***************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
***************************************************************************************************************************************************

Processing teensy40 (platform: teensy; board: teensy40; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing teensy
Downloading  [####################################]  100%
Platform Manager: teensy @ 4.12.0 has been installed!
The platform 'teensy' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/framework-arduinoteensy @ ~1.153.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoteensy @ 1.153.0 has been installed!
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.50401.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-gccarmnoneeabi @ 1.50401.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~2.20501.7
Tool Manager: tool-scons @ 2.20501.191222 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.12.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy 1.153.0 (1.53)
 - toolchain-gccarmnoneeabi 1.50401.0 (5.4.1)
Converting Blink.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 89 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/teensy40/src/Blink.ino.cpp.o
Compiling .pio/build/teensy40/FrameworkArduino/AudioStream.cpp.o
Compiling .pio/build/teensy40/FrameworkArduino/Blink.cc.o
In file included from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:45:0,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from /tmp/tmpgHWArd:1:
/home/dean/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
Compiling .pio/build/teensy40/FrameworkArduino/EventResponder.cpp.o
*** [.pio/build/teensy40/src/Blink.ino.cpp.o] Error 1
In file included from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:45:0,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/AudioStream.cpp:32:
/home/dean/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
*** [.pio/build/teensy40/FrameworkArduino/AudioStream.cpp.o] Error 1
In file included from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:45:0,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/Blink.cc:3:
/home/dean/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
*** [.pio/build/teensy40/FrameworkArduino/Blink.cc.o] Error 1
In file included from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/wiring.h:45:0,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/WProgram.h:45,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/Arduino.h:6,
                 from /home/dean/.platformio/packages/framework-arduinoteensy/cores/teensy4/EventResponder.cpp:34:
/home/dean/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/5.4.1/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
*** [.pio/build/teensy40/FrameworkArduino/EventResponder.cpp.o] Error 1
=========================================================== [FAILED] Took 50.40 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
teensy40       FAILED    00:00:50.404
====================================================== 1 failed, 0 succeeded in 00:00:50.404 ======================================================
```

maxgerhardt, 2021-01-07:
It was uploaded to https://bintray.com/platformio/tool-packages/toolchain-gccarmnoneeabi-linux_aarch64#files as `5966cc5-toolchain-gccarmnoneeabi-linux_aarch64-1.90301.200702.tar.gz`, but the teensy platform manifest still references the older version, so you'll have to do 

``` 
platform_packages = 
   toolchain-gccarmnoneeabi@1.90301.200702
```
in the `platformio.ini`
which should then pull it from the official PlatformIO source.

blackketter, 2021-01-07:
Thanks @maxgerhardt .

Got a little further along but failed with: 
```
Linking .pio/build/teensy40/firmware.elf
/home/dean/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -larm_cortexM7lfsp_math
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy40/firmware.elf] Error 1
```

maxgerhardt, 2021-01-07:
>/home/dean/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -larm_cortexM7lfsp_math 

Well then the uploaded toolchain did not have the `libarm_cortexM7lfsp_math.a` after all, fail. 

@valeros you need to update the toolchain again :D

bornae, 2021-01-08:
@valeros With your updates, I am able to build and upload to a teensy 4.0 with a basic platformio.ini file. This was done on an Nvidia Jetson Xavier NX

The only thing I had to do manually was installing `sudo apt install libusb-0.1-4`

Full output below:

@maxgerhardt Teensyduino uses gccarmnoneeabi version 5.4.1. Not sure why you were saying that is the wrong version. 

platformio.ini:
```ini
[env:teensy40]
platform = teensy
framework = arduino
board = teensy40
targets = upload
upload_protocol = teensy-cli
```

Full build output:
``` bash
bemami@jetson-nx:~/code$ rm -rf ~/.platformio/
bemami@jetson-nx:~/code$ ./deploy
Generating Teensy Sketch Folder =============
Done!
***********************************************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
***********************************************************************************************************************************************************************************************************************************************************

Processing teensy40 (platform: teensy; framework: arduino; board: teensy40)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing teensy
Downloading  [####################################]  100%
Platform Manager: teensy @ 4.12.0 has been installed!
The platform 'teensy' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.50401.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-gccarmnoneeabi @ 1.50401.210106 has been installed!
Tool Manager: Installing platformio/framework-arduinoteensy @ ~1.153.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoteensy @ 1.153.0 has been installed!
Tool Manager: Installing platformio/tool-teensy @ <2
Unpacking  [####################################]  100%
Tool Manager: tool-teensy @ 1.152.200516 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40001.0
Tool Manager: tool-scons @ 4.40001.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.12.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy 1.153.0 (1.53)
 - tool-teensy 1.152.200516 (1.52)
 - toolchain-gccarmnoneeabi 1.50401.210106 (5.4.1)
Converting src.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 89 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/teensy40/src/src.ino.cpp.o
Linking .pio/build/teensy40/firmware.elf
Checking size .pio/build/teensy40/firmware.elf
Building .pio/build/teensy40/firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.8% (used 82612 bytes from 524288 bytes)
Flash: [          ]   2.9% (used 59008 bytes from 2031616 bytes)
Configuring upload protocol...
AVAILABLE: jlink, teensy-cli, teensy-gui
CURRENT: upload_protocol = teensy-cli
Rebooting...
Uploading .pio/build/teensy40/firmware.hex
Teensy Loader, Command Line, Version 2.2
Read ".pio/build/teensy40/firmware.hex": 59020 bytes, 2.9% usage
Error opening USB device: No error
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read ".pio/build/teensy40/firmware.hex": 59020 bytes, 2.9% usage
Programming.......................................................
Booting
============================================================================================================== [SUCCESS] Took 39.53 seconds ==============================================================================================================

Environment    Status    Duration
-------------  --------  ------------
teensy40       SUCCESS   00:00:39.528
=============================================================================================================== 1 succeeded in 00:00:39.528 ===============================================================================================================
```

maxgerhardt, 2021-01-08:
@bornae  You're right, I was used to the 9.x GCC version but I should have looked at the upload date and see that `1.50401.210106` version is the right one. Corrected. Good that it works!

John-Wilcox, 2021-02-15:
The repository https://bintray.com/platformio/dl-packages no longer exists.  
Is that because of this? https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/  
Is there another workaround for compiling Teensy 4.0 using PlatformIO on a Raspberry Pi?

maxgerhardt, 2021-02-15:
Well the repo is gone (see https://github.com/platformio/platformio-core/issues/3836) but that is only one of the content mirrors, so PlatformIO still works. Per linked issue you can also search it. What exact package do you need, or, what raspbi do you have? 

If it's running an aarch64 OS version you should still be able to do 

```
[env:teensy40]
platform = teensy
framework = arduino
board = teensy40 
platform_packages = 
   toolchain-gccarmnoneeabi@https://dl.registry.platformio.org/download/platformio/tool/toolchain-gccarmnoneeabi/1.90301.200702/toolchain-gccarmnoneeabi-linux_aarch64-1.90301.200702.tar.gz
   tool-teensy@https://github.com/maxgerhardt/pio-tool-teensy-arm64/archive/master.zip
```
along with the lib file fix as mentioned in https://github.com/platformio/platform-teensy/issues/51#issuecomment-679424383.

John-Wilcox, 2021-02-15:
@maxgerhardt ;   thank you for the quick response.   I'm on a Rpi 4 using Rpi OS Lite (32-bit)
looking for toolchain-gccarmnoneeabi-linux_armv6l-1.90301.200702.tar.gz

John-Wilcox, 2021-02-15:
Found it

https://dl.registry.platformio.org/download/platformio/tool/toolchain-gccarmnoneeabi/1.90301.200702/toolchain-gccarmnoneeabi-linux_armv6l-1.90301.200702.tar.gz
Thank you for your help

nuwaveit, 2021-02-18:
Thanks everyone for these posts.  I just switched from Arduino IDE to PlatformIO (VS Code) on my Raspberri Pi 400, running 32-bit Raspbian desktop...and ran into the same issue.  Thought I would post in case anyone was searching...

add this to the platformio.ini:
platform_packages = 
   toolchain-gccarmnoneeabi@~1.90301.0
   tool-teensy@https://github.com/maxgerhardt/pio-tool-teensy-arm/archive/master.zip

grabbed the lib from:
https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Lib/GCC/libarm_cortexM7lfsp_math.a

and placed the lib here:
.pio/build/teensy40

Builds great.  Took like 30 minutes to download the tools, tho...

stawiski, 2021-05-02:
My error when building on RPI4 was:
```
/home/pi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -larm_cortexM7lfsp_math
collect2: error: ld returned 1 exit status
```

And to solve it the `libarm_cortexM7lfsp_math.a` must be put into
```
~/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/
```

valeros, 2021-11-04:
@maxgerhardt Many thanks, now we have a complete set for ARM SBCs.

closed 2021-11-04 by valeros