platformio/platform-teensy #54: Update to Teensyduino 1.48

blackketter, 2019-11-07:
Teensyduino has been updated to 1.48 with substantial improvements for Teensy 4.0.  Would be great to pull that in.  Thanks!

blackketter, 2019-11-11:
Note that the teensy-cli uploader ( https://github.com/PaulStoffregen/teensy_loader_cli )  now works with teensy40 and should be updated as such along with the 1.48 release.

ivankravets, 2019-11-11:
@PaulStoffregen could you help us with binaries for "teensy_loader_cli" for macOS, Linux (32/64/arm), Windows? 
Thanks!

ivankravets, 2019-11-12:
1. Teensyduino 1.48 is out, see https://community.platformio.org/t/teensy-dev-platform-4-6-0/10514

2. I updated this issue to "Update Teensy Loader CLI". So, let's wait when someone prepares binaries for `teensy_loader_cli` for macOS, Linux, Windows.

vshymanskyy, 2020-01-23:
PIO `teensy-cli` tool cannot upload to `Teensy 4.0` due to this.

ivankravets, 2020-01-23:
`teensy_loader_cli` has been updated for macOS and Linux x64. Now, the only Window CLI is outdated.

ivankravets, 2020-01-24:
`teensy_loader_cli` was also updated for Windows. 

Thanks, @valeros, @vshymanskyy for the binaries!

closed 2020-01-24 by ivankravets

PaulStoffregen, 2020-01-24:
@ivankravets - Soon Teensy 4.0 will begin uses a -DARDUINO_TEENSY40 define, similar to the convention Arduino uses for their boards.  Details here:

https://forum.pjrc.com/threads/58028?p=227510&viewfull=1#post227510

Recommend you add this to the flags used for Teensy compile, in prep for the next core lib update.

ivankravets, 2020-01-24:
@PaulStoffregen thanks for the info! Should we also add it to other boards? For example, `-D ARDUINO_TEENSY30`? 

@valeros please add for all Teensy boards `-D ARDUINO_***` macro, we have it for other dev-platforms (ESP32, ESP8266, AVR, Atmel SAM, etc)

PaulStoffregen, 2020-01-24:
> Should we also add it to other boards?

Yes, though it probably will only matter with Teensy 4.x.

Probably a good idea for non-Teensy boards.  But important to check their boards.txt file, so you get the exact same "build.board" string each board defines for use in Arduino.