platformio/platform-teensy #58: Update to Teensy 1.50

blackketter, 2020-02-01:
See https://github.com/PaulStoffregen/cores/releases/tag/1.50

Thanks!

blackketter, 2020-02-22:
Whoops, 1.51 is out now:  https://github.com/PaulStoffregen/cores/releases/1.51

Theremingenieur, 2020-03-03:
It's a little unsatisfying... The latest Teensyduino releases are huge steps forward and fix lots of issues, especially around the Teensy 4. When 1.49 came out on Jan 14th, we got the PlatformIO update 9 days later. 1.50 came out on Feb 1st, and we haven't seen an update for 32 days and still pending. 1.51 came out on Feb 22nd, 10 days ago. PJRC is already working on the 1.52 beta for 8 days now. I don't know who is being PlatformIO, but I'd suggest that they read the PJRC forum from time to time, follow the beta publications and be prepared at the release date.

blackketter, 2020-03-03:
Any idea when this might happen, @valeros & @ivankravets ?

ivankravets, 2020-03-03:
You can subscribe for releases here https://github.com/PaulStoffregen/cores/releases

We will fix this issue soon.

Theremingenieur, 2020-03-03:
Thank you for the quick reply! Much appreciated!

valeros, 2020-03-04:
Packages have been updated, please retest with the upstream version http://docs.platformio.org/en/latest/platforms/teensy.html#upstream

blackketter, 2020-03-04:
Initial testing seems ok.  Thanks!

Theremingenieur, 2020-03-04:
Update worked. Thank you!

closed 2020-03-04 by ivankravets