platformio/platform-teensy #61: Toolchain-gccarmnoneeabi new versions :”cannot find -larm_cortexM4l_mat”

FiratBerkCakar, 2020-03-26:
Hello everyone,
I am writing something for Teensy 3.2 and wanted to use some c++ constexper functionalities offered by cpp17. The default toolchain-gccarmnoneeabi ( version 1.50401.190816) doesnt seem to support some of the stuff i needed, so i decided to compile with a newer version using the build flag
“platform_packages = toolchain-gccarmnoneeabi…” but i get the following error when i try to compile :

`…arm-none-eabi/bin/ld: cannot find -larm_cortexM4l_math
`

for the following versions:

* toolchain-gccarmnoneeabi-linux_x86_64-1.70201.0.tar.gz

* toolchain-gccarmnoneeabi-linux_x86_64-1.80201.181220.tar.gz

I solved this issue by simply copying the missing library form the default version but this seemed like an ugly hack since to my understanding it should be there in the first place.

I just wanted to bring this issue to public attention and ask if i am doing something wrong that causes this issue rather than a package simple missing a library.

PS: I am using Linux so some of the windows versions might not have this issue.

valeros, 2020-03-26:
`~1.50401.0` is default version for `teensy`, there are no guarantees that other toolchain packages are compatible with `teensy` platform.

closed 2020-03-26 by valeros

cinderblock, 2021-11-11:
@FiratBerkCakar I too am looking for how to do this as well as I also want to use some newer GCC features. The PIO docs suggest this [as how to get the latest toolchain](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages). It's a bummer than this one file is missing.

@valeros What would it take to bring the Teensy platform up to modern tools?

PaulStoffregen, 2021-11-11:
I do not recommend using a newer toolchain or C++17 dialect with Teensy.

We test all the Teensyduino libraries extensively with C++14 dialect and a specific toolchain which is known to produce the fastest compiled code.  So far, all newer versions have resulted in slightly slower code optimization.

If you change to a newer toolchain or compile with substantially different compiler options than we have used to test all the libraries, odds are strong you will run into problems.

cinderblock, 2021-11-12:
Hey @PaulStoffregen, I hear that concern and it's a good one.

However, for me, the "`warning: '[struct]' is too small to hold all values of 'enum class [...]'`" messages (that [**cannot**](https://stackoverflow.com/a/36030403/4612476) be disabled) are becoming more of a problem than anything else.

Case in point: I just spent a day tracking down a new bug. Had I not had pages of unnecessary warnings about enum sizes, I might have noticed the warning telling me about the issue sooner. (Granted, `-fpermissive`, enabled by default, was also not helping).

I fortunately have a pretty comprehensive set of tests for my hardware and everything seems to work fine so far. I'm going to be moving forward with GCC 9.3.1 but I know it's on me to test everything.

PS, it seems GCC 9.3.1 (with `-std=gnu++14`) produces smaller (and, therefore, *presumably* more optimized assembly) for my particular application:

#### `toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)`

```log
RAM:   [=         ]  12.9% (used 8468 bytes from 65536 bytes)
Flash: [==        ]  20.3% (used 53260 bytes from 262144 bytes)
```

#### `toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)`

```log
RAM:   [=         ]  12.4% (used 8152 bytes from 65536 bytes)
Flash: [==        ]  19.9% (used 52128 bytes from 262144 bytes)
```