platformio/platform-teensy #62: Support for Teensy 4.1

sstaub, 2020-05-11:
Can you add Teensy 4.1 ? It was just released.

blackketter, 2020-05-11:
This will probably need to wait until Teensyduino 1.52 is released, which should be in the next couple of days.

sstaub, 2020-05-11:
Yes, you are right. I have ordered mine today and will receive the board the next days.

sstaub, 2020-05-16:
Teensyduino 1.52 is released

davidtopham, 2020-05-16:
I'd also like this - I've got a Teensy 4.1 that's just arrived sat waiting for this support.

I posted a question on the forum about framework-arduinoteensy, as the packaging of that seems like quite an opaque process, whereas platform-teensy is in an open repository (here!).
https://community.platformio.org/t/how-is-framework-arduinoteensy-packaged/13710

I'd like to be able to help get this working, but can't find a way of getting framework-arduinoteensy updated without someone on the PlatformIO team doing it.

closed 2020-05-16 by valeros

valeros, 2020-05-16:
I've added Teensy 4.1, please retest with the upstream version http://docs.platformio.org/en/latest/platforms/teensy.html#upstream

sstaub, 2020-05-17:
Thank you, my  4.1 is hanging on the mailing route, I hope I can test on monday.

davidtopham, 2020-05-17:
Thanks, that seems to work. I think the 'maximum_size' line in teensy41.json might be wrong though, the boards.txt file in Teensyduino 1.52 suggests it should be 8126464.

I did need to manually specify to use version 1.52 of tool-teensy with:
platform_packages = tool-teensy@1.152
For some reason it still picks up the 1.51 version of the teensy loader tools.

mstaack, 2020-05-19:
working perfectly with bleeding edge on a teensy 4.1:
```
[env:teensy41]
platform = https://github.com/platformio/platform-teensy.git
board = teensy41
framework = arduino
build_flags = -D USB_MIDI_AUDIO_SERIAL
```

sstaub, 2020-05-20:
Works, thank you.

commit dd0da0be (2020-05-16): Add initial support for Teensy 4.1 // Resolve #62