platformio/platform-teensy #69: USB serial not working on Teensy 4.0 aftert upgrade from 4.70 to 4.11.0

torsten-pf, 2020-11-24:
Hi,

today I upgraded my current project from 4.7.0 (Teensy core 1..49) to 4.11.0 (1.53) and the USB serial interface stopped working (message about unknown device - code 43 on Windows - no descriptors are available for device). I assumed this was issued by my code changes, but now I'm sure that it was the update. After downgrading the project to 4.7.0 it is working again as before.

Do I need to add build_flags for the new version? Here is my platform.ini:

```
[env]
build_flags = !python svnrev.py, -felide-constructors, -std=c++0x

[env:teensy40@4.7.0]
platform = teensy@4.7.0
framework = arduino
board = teensy40

[env:teensy40@current]
platform = teensy
framework = arduino
board = teensy40
```
The verbose build shows a difference in a define during build: 4.7.0 uses `-DTEENSY40` while 4.11.0 applies `-DARDUINO_TEENSY40`. Here are two compile statements from the build:

* 4.7.0
```
arm-none-eabi-gcc -o .pio\build\teensy40\FrameworkArduino\usb_touch.c.o -c -felide-constructors -Wall -ffunction-sections -fdata-sections -mthumb -mcpu=cortex-m7 -nostdlib -fsingle-precision-constant 
-mfloat-abi=hard -mfpu=fpv5-d16 -O2 -DPLATFORMIO=50003 -D__IMXRT1062__ -DTEENSY40 -DSVN_REV=232 -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=149 -DCORE_TEENSY -DF_CPU=600000000 -DLAYOUT_US_ENGLISH 
-I<HOMEDIR>\.platformio\packages\framework-arduinoteensy@1.149.0\cores\teensy4 <HOMEDIR>\.platformio\packages\framework-arduinoteensy@1.149.0\cores\teensy4\usb_touch.c
```

* 4.11.0

```
arm-none-eabi-gcc -o .pio\build\teensy40@current\FrameworkArduino\usb_touch.c.o -c -felide-constructors -Wall -ffunction-sections -fdata-sections -mthumb -mcpu=cortex-m7 -nostdlib -fsingle-precision-constant 
-mfloat-abi=hard -mfpu=fpv5-d16 -O2 -DPLATFORMIO=50003 -D__IMXRT1062__ -DARDUINO_TEENSY40 -DSVN_REV=232 -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=153 -DCORE_TEENSY -DF_CPU=600000000 -DLAYOUT_US_ENGLISH 
-I<HOMEDIR>\.platformio\packages\framework-arduinoteensy\cores\teensy4 <HOMEDIR>\.platformio\packages\framework-arduinoteensy\cores\teensy4\usb_touch.c
```

Any help is appreciated.

valeros, 2020-11-27:
Hi @torsten-pf ! Besides switching to new macro names, we also updated Teensyduino to the latest v1.53. Could you please try to run `teensy40@current` environment with `-DTEENSY40` macro in your build flags? Also, is there any chance you could run your project in Arduino IDE with the latest  Teensyduino? Thanks

torsten-pf, 2020-11-27:
Hi @valeros,

tried it right now with the current version and `-DTEENSY40`. But without success (same result as before). Will try the Arduino way now.

torsten-pf, 2020-11-27:
@valeros, Arduino 1.8.13 with Teensyduino 1.53 has the same problem. Will ask in the teensy forum.

torsten-pf, 2020-11-27:
@valeros, I found the problem (it is most of the time sitting in front of the computer...). I used an old version of a library (T4_PowerButton) that caused this error. I'll close this issue now.

closed 2020-11-27 by torsten-pf