platformio/platform-teensy #70: Blink Example setup/loop not working as part of larger project (for Teensy 3.1)

chlirre, 2021-01-08:
## Background 
I am currently attempting to port the [firmware for synth module Ornament & Crime](https://ornament-and-cri.me/firmware/) to PlatformIO to be able to develop under later OS X versions.

In addition to being written for Teensy 3.1, the O & C project has some restrictions in terms of compiler.
> gcc 4.8

Which is enforced by [stating the toolchain-gccarmnoneeabi version](https://github.com/chlirre/O_C-HemisphereSuite/blob/blink-loop/software/platformio.ini#L13).

I am specifically porting the alternative firmware [Hemisphere](https://github.com/Chysn/O_C-HemisphereSuite ).

I'm on OS X 10.13.2, running PlatformIO IDE (with _PlatformIO Core, version 5.0.4_).

I tried the suggestions found at https://github.com/platformio/platformio-core/issues/566, without success. I tried prepending As & Zs to the main file without luck.

I cannot tell if the issue lies within PlatformIO, this platform or elsewhere. All I know is https://github.com/Chysn/O_C-HemisphereSuite builds, uploads and runs fine with the Arduino IDE and that I've made as little changes as possible to that code.

## The issue
The project is in a state where everything builds just fine (apart from some warnings).
Uploading/programming (with both teensy-cli and teensy-gui) seens to work just fine.

But beyond that point, **there is no indication of the firmware running**. 

- As in no serial output. 
- Nothing showing on the screen.
- And in the case of borrowing the "main/loop" file from the led blink example, **the LED is not blinking**.

#### Not working (not blinking)
##### Loop/setup
https://github.com/chlirre/O_C-HemisphereSuite/blob/blink-loop/software/src/Main.cpp
Note that the the only difference is the `OC::CORE::ticks` initialization which I had to leave in there to avoid linker errors.

##### Env/platform
https://github.com/chlirre/O_C-HemisphereSuite/blob/blink-loop/software/platformio.ini

#### Working (blinking)
##### Loop/setup
https://github.com/chlirre/platform-teensy/blob/o_c-platform/examples/arduino-blink/src/Blink.cpp

##### Env/platform
https://github.com/chlirre/platform-teensy/blob/o_c-platform/examples/arduino-blink/platformio.ini

maxgerhardt, 2021-01-09:
Cross-posting reference: https://community.platformio.org/t/teensy-3-1-issue-firmware-uploading-but-no-indication-of-running/18398

valeros, 2021-01-11:
Hi @chlirre ! There might be plenty of reasons. First, you need to make sure that the package versions are identical (toolchain, framework, libs, etc). I also noticed you use `-O2` optimization. Have you tried other levels?

BTW, this line `toolchain-gccarmnoneeabi@1.40804` seems invalid to me, probably should be something like `toolchain-gccarmnoneeabi@~1.40804.0`

chlirre, 2021-01-21:
@valeros I tried a few different optimizations, but the instructions for compiling the original project explicitly state `4) select Faster (= o2) in Tools > Optimize (teensyduino 1.34 and 1.35)` (https://ornament-and-cri.me/firmware/) so I stuck with that.

valeros, 2021-01-21:
Could you please try to compile your project in verbose mode and attach the log here?