platformio/platform-teensy #72: Teensy project cannot use arduino library

HYC23, 2021-03-20:
I'm using platform io to build my teensy 3.1 project. It worked great on my desktop at my company. However, I try to build on my laptop, it's giving an error
fatal error: Wire.h: No such file or directory
All other Arduino libraries cannot find as well. When I build an Arduino project (atmelavr), everything works fine.

My platformio.ini
[env:teensy31]
platform = teensy
board = teensy31
framework = arduino

I also tried to add 
lib_deps = Wire
Nothing changed.

Any ideas what's wrong with my setup? 

Thanks for your help.

blackketter, 2021-03-20:
What OS are you using for your build and what's the source file that you are trying to build?

HYC23, 2021-03-21:
I'm using windows with the latest platform io core and latest vs code.

I tried different source files, and any one of them include Wire.h or SPI.h can not build, because it cannot find the library.
eg.

#include <Arduino.h>
#include <Wire.h>
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

shows error:
src\main.cpp:2:18: fatal error: Wire.h: No such file or directory

I tried to reinstall the teensy board and platform io, still have this error.

Strangely, in my company windows computer with the same setup works fine.

any ideas?

blackketter, 2021-03-21:
Try renaming the file to main.ino.

If that doesn't work, I'm at a loss.  No windows around here.

HYC23, 2021-03-21:
Change the name to .ino does not work.
But I figure it out. Delete the .platformio folder under user/ will solve this problem.

closed 2021-03-21 by HYC23