platformio/platform-teensy #73: Teensy 3.6 uploads through PlatformIO resulting in unrecognized USB device.

CapeC0d3r, 2021-03-25:
I was told to log an issue here after posting this  to the PIO Forum: 
https://community.platformio.org/t/teensy-3-6-upload-issues-upload-fails-usb-com-port-disappears-code-43/20240/3

I am hoping sharing a link to that post is good enough, and that I don't need to rewrite all of that. 

I am able to recreate the issue described in that post while using PIO to upload a hex file to a teensy 3.6 through a USB hub, but I can't recreate the problem while uploading the same hex file to the same teensy over the same USB hub.  If I plug into another hub (same model, different side of my desk) I am not able to recreate the issue.  Also unable to recreate the problem if plugged directly to a USB port on the from of my PC.  

Both hubs are:
https://www.amazon.com/Splitter-Charging-Individual-Switches-Extension/dp/B07G8CMR18/ref=sr_1_1_sspa?crid=25Q2M2KL96Q6P&dchild=1&keywords=atolla+usb+hub+3.0+powered&qid=1616694827&sprefix=atolla+%2Caps%2C171&sr=8-1-spons&psc=1&smid=A2KVFO4XQ2RAFL&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUE1MVVFVVlDOEpZNDkmZW5jcnlwdGVkSWQ9QTA2MDU3MzEzVDYzN00zSjZPTkI4JmVuY3J5cHRlZEFkSWQ9QTA0NjM2MjQzTkNaQUhWWEpVVVpRJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

Here are some Teensyloader (1.52) logs I took while attempting to reproduce the problem using different USB connections.  

Using the Teensyloader GUI to manually upload a hex file built using PIO.  No problem:
[T36_1_Teensyloader_GUI_Multiple_Successful_Uploads_No_Problem.txt](https://github.com/platformio/platform-teensy/files/6206804/T36_1_Teensyloader_GUI_Multiple_Successful_Uploads_No_Problem.txt)

Using PIO to upload through a USB 3.0 port on the front of my PC:
[T36_1_PIO_Upload_Motherboard_USB3_Port_No_Problem.txt](https://github.com/platformio/platform-teensy/files/6206807/T36_1_PIO_Upload_Motherboard_USB3_Port_No_Problem.txt)

Using PIO to upload through one of my USB 3.0 hubs (hub A) with problem:
[T36_1_PIO_Upload_USB_Hub_A_Problem.txt](https://github.com/platformio/platform-teensy/files/6206817/T36_1_PIO_Upload_USB_Hub_A_Problem.txt)

[T36_1_PIO_Upload_USB_Hub_A_Second_Example.txt](https://github.com/platformio/platform-teensy/files/6206819/T36_1_PIO_Upload_USB_Hub_A_Second_Example.txt)

Using PIO to upload through a different USB 3.0 hub (hub b) without problems: 
[T36_1_PIO_Upload_USB_Hub_B_Multiple_Times_With_No_Problem.txt](https://github.com/platformio/platform-teensy/files/6206941/T36_1_PIO_Upload_USB_Hub_B_Multiple_Times_With_No_Problem.txt)

If there is anything else I can provide you with don't hesitate to reach out.  I am easily able to recreate the problem while using "USB Hub A"  

If the answer is simply "don't use that USB hub" that is fine, but my friend is also experiencing the same thing on his PC while connected to a USB hub built into his monitor.  Since I have suggested that he build his code in PIO and then upload it manually through the Teensyloader GUI he has not experienced the problem using the same USB connection.

Here is a simple PIO project I am loading to troubleshoot the issue: 
[Hello_Teensy.zip](https://github.com/platformio/platform-teensy/files/6206878/Hello_Teensy.zip)

And just incase you need it my teensy "package" from "user\.platformio\packages":
[tool-teensy.zip](https://github.com/platformio/platform-teensy/files/6206885/tool-teensy.zip)

CapeC0d3r, 2021-03-25:
Two things: 

I just referred back to my PIO post and noticed that am calling the problematic USB hub "hub B" there.  Sorry for any confusion that causes.  Moral of the story is one seems to work fine and the other does not.  

> 13:35:05.302 (reboot 3): Status: 1, 1, 0, 0, 0, 0, C:\Users\email\Documents\PlatformIO\Projects\Hello_Teensy\.pio\build\teensy36\, firmware.hex

I am calling the entries that look like this in the middle of block transfers "reboots" in my post, but I'm pretty sure they are just status requests and the "(reboot 3)" is just some sort of identifier.  I don't really know what any of the information in those logs actually means, I just know that the upload logs generated during a PIO update look a lot different than the upload logs using the teensyloader gui application by itself.

valeros, 2021-03-26:
Hi @CapeC0d3r ! Unfortunately, I'm not sure we can help you here. Is there any chance that you can try the same configuration on a Linux machine? Have you tried to compile and upload the same project using Arduino IDE? Does it work reliably there? Also, have you tried the CLI version of the Teensy loader, e.g.:
```
[env:teensy36]
platform = teensy
framework = arduino
board = teensy36
upload_protocol = teensy-cli
```

CapeC0d3r, 2021-03-26:
@valeros 

> Is there any chance that you can try the same configuration on a Linux machine?

Sure, I have been meaning to spin up a linux development VM with PIO.  I spent a good amount of time banging my head against this yesterday, and I have a few things I need to get done.  If I finish up I will get setup on Linux, but I don't expect I will see a similar issue.  I don't know what's going on exactly, but judging by what I have read and seen, I think it is a windows issue.  That said, I want to reiterate that I can't recreate the issue if I use the teensyloader.exe gui program by itself, I see totally different log activity, and although I am totally ignorant as to what is going on between PIO and the teensyloader gui, the logs from teensyloader when being used by itself look a lot "cleaner".  I was surprised to see the differences in logs when I had the teensy-gui option selected in my .ini file, BUT again, I don't really know how it works under the hood.  If anyone was to look into the issue, I'd look into any differences between the upload process of someone using teensyloader gui by itself on windows, and teensy-gui option through PIO.  

I also want to mention that now that I am fairly certain this has nothing to do with my code, or the teensyloader's ability to load the program, I am not very concerned about the issue.  I just wanted to post my information for anyone else that is losing sleep over it.  PERSONALLY I am totally fine with not using the problematic USB connection and moving on knowing that the physical connection to a windows machine can be quirky while uploading from PIO using the teensy-gui option. 

> Have you tried to compile and upload the same project using Arduino IDE? 

I have not.  I will give that a shot when I have some free time. 

> Also, have you tried the CLI version of the Teensy loader

This is on my to-do list.  I will see if I get the same issue after I get it working.

Thanks for the reply, and for looking through my rambling posts.

CapeC0d3r, 2021-04-01:
Just following up after a few days of uploading code without a problem using PIO over a different USB connection. 

Paul also asked if I was able to make it happen when I linked this post in the PJRC forum, and I finally got around to installing the latest Arduino IDE and Teensyduino, and then trying to recreated the problem using the Arduino IDE to load the firmware.  I was able to get it happen after 4-5 back to back uploads.  

Teensyloader Log: 
[Arduino_IDE_Code_43_Teensyloader_Log.txt](https://github.com/platformio/platform-teensy/files/6243786/Arduino_IDE_Code_43_Teensyloader_Log.txt)

Like I mentioned before, I am unable to get this to happen over a different USB connection, so I'm not really looking for a fix.  Feel free to close the issue if you want.  I will reiterate that my friend has also experienced this over a USB hub built into his monitor.  

The PJRC Troubleshooting Tip Page says this 

> Some USB hubs have trouble handling the rapid sequence of connect/disconnect events when programming Teensy. Most hubs work fine, but the few that do not can cause very strange problems. If you experience trouble, always try connecting Teensy directly with a quality USB cable. 

That's good enough for me, I just wanted to write my experience down to help anyone else who might be stressing out about it.

valeros, 2021-04-01:
Thanks for sharing your experience. Hope this topic will save someone from hours of googling.

closed 2021-04-01 by valeros