platformio/platform-teensy #75: Please update teensy_loader_cli

jcw, 2021-08-01:
The source code for this is at: https://github.com/PaulStoffregen/teensy_loader_cli
It now has support for TEENSY41, the current PIO setup only goes to TEENSY40. This can be checked with:

    ~/.platformio/packages/tool-teensy/teensy_loader_cli --list-mcus

But ... for MacOS, _please_ build with LIBUSB defined in the Makefile, don't use IOKit. The reason for this is that the libusb version is able to get the Teensy into boot mode with "-s", whereas IOKit does not support that feature.

As far as I understand it (which says little), the only way to get a Teensy back to a normal running state from either serial or boot mode, is to use this command:

    teensy_loader_cli --mcu=imxrt1062 -s -b

For some background, see [my post on the Teensy forum](https://forum.pjrc.com/threads/67851-Launching-RAM-based-applications?p=285042&viewfull=1#post285042), as I try to implement a streamlined RAM-based dev cycle.

valeros, 2021-08-04:
Hi @jcw ! I've just published a new `tool-teensy` package for MacOS with the latest `teensy_loader_cli` inside. Could you please run `pio platform update teensy` and confirm that it works as expected?

jcw, 2021-08-04:
Yep, update works like a charm, "-s" now works on MacOS - many thanks!

closed 2021-08-09 by valeros