platformio/platform-teensy #76: Any potential issues with Teensy 4.12 and using Teensyduino 1.54?

m-r-m-s, 2021-08-06:
Hi, 

I have Teensyduino 1.54 installed alongside my Arduino IDE 1.8.15. I have a sketch that does no longer works as expected which I thought was due to the new Teensyduino version. 

But perhaps it is the Arduino IDE that is the issue? The sketch I am using is based on an example from this library: https://github.com/midilab/uClock

However, when I compile the same sketch in PlatformIO using Teensy platform 4.12 using the same Teensyduino 1.54, the sketch works as expected. 

I am concerned if I update to 4.13 on PlatformIO the sketch won't work there either. Is there a way to revert back to 4.12 if that is the case?

valeros, 2021-08-06:
Hi @m-r-m-s ! If it doesn't work in the Arduino IDE then the issue is most likely hidden somewhere in the framework. It's safe to update to the latest v4.13 and check if it works with PlatformIO, as it's always possible to rollback to the previous version of the platform:
```ini
[env:teensy41]
platform = teensy@~4.12.0
framework = arduino
board = teensy41 
```

closed 2021-08-06 by valeros

m-r-m-s, 2021-08-06:
> Hi @m-r-m-s ! If it doesn't work in the Arduino IDE then the issue is most likely hidden somewhere in the framework. It's safe to update to the latest v4.13 and check if it works with PlatformIO, as [...]

Ok, Thank you. I'll investigate this further. I have updated to 4.13 and the sketch continues to work with no issue. This is what is displayed in the platformio.ini file:
```
[env:teensylc]
platform = teensy
board = teensylc
framework = arduino
lib_deps =
	midilab/uClock@^0.10.4
	olikraus/U8g2@^2.28.8
```