platformio/platform-teensy #81: IntelliSense not opening certain avr header files on empty teensy 2.0 project

JesseRiemens, 2021-11-05:
Hi, I came across this issue: [Discussion on PIO Community](https://community.platformio.org/t/teensy-2-0-empty-project-has-include-errors/23985/16)

What's happening: I am creating a clean teensy 2.0 project and IntelliSense gives this error:  [image]

The project does compile but IntelliSense refuses to work. As worked out in the community thread, I have all extensions disabled, the files IntelliSense needs are there (it compiles) and I am on these versions:

-  VSCode 1.61.1
- piohome 3.3.4
- piocore 5.2.1
- extension 2.3.3
- Platform teensy
    - Updating platformio/teensy                    4.13.1                             [Up-to-date]
    - Updating platformio/toolchain-atmelavr        1.50400.190710 @ ~1.50400.0        [Up-to-date]
    - Updating platformio/toolchain-gccarmnoneeabi  1.50401.190816 @ ~1.50401.0        [Up-to-date]
    - Updating platformio/framework-arduinoteensy   1.154.0 @ ~1.154.0                 [Up-to-date]
    - Updating platformio/tool-teensy               1.155.0 @ <2                       [Up-to-date]

My c_cpp_properties.json looks [like this.](https://pastebin.com/w6u52SSS) 
Does anybody have a clue?

valeros, 2021-11-09:
Hi @JesseRiemens! It's strange, but I cannot reproduce the issue with the latest platform v4.14. Could you please pull the latest release via `pio platform update teensy` and try again?

I also noticed that your ` c_cpp_properties.json` and the error message on the screenshot refer to different disk drivers, any chance you're using network drive or something unusual?

JesseRiemens, 2022-03-25:
Hi, I wanted to revive this issue since I did a clean windows reinstall and found the same problem, with a different board also using the arduino framework. I tried creating a project on the C (SSD) disk, and on the Z (HDD) disk but that made no difference. Anyone who can help me here?

JesseRiemens, 2022-10-24:
Reset my global settings.json and it fixed it!

closed 2022-10-24 by JesseRiemens