platformio/platform-teensy #82: cores/teensy/Blink.cc causes build failure for some machines

maxgerhardt, 2021-12-21:
With the very latest teensy platform 4.14.0, the used `framework-arduinoteensy` (1.155.0) has the file `cores/teensy4/Blink.cc` which contains

```cpp
// Only for testing with the Makefile - not actually part of the core library

#include <Arduino.h>

int led = 13;

void setup() {
  pinMode(led, OUTPUT);
}

void loop() {
  digitalWrite(led, HIGH);
  delay(1000);
  digitalWrite(led, LOW);
  delay(1000);
}
```

Which, if compiled, conflicts with user-implemented `setup()` and `loop()` functions

This file is included in the build for a standard

```ini
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
```
configuration, as evident by the build log

```
Compiling .pio\build\teensy41\FrameworkArduino\Blink.cc.o
```

However, the build still passes for me on my machine

```
Linking .pio\build\teensy40\firmware.elf
Calculating size .pio\build\teensy40\firmware.elf
Checking size .pio\build\teensy40\firmware.elf
Building .pio\build\teensy40\firmware.hex
teensy_size: Memory Usage on Teensy 4.0:
teensy_size:   FLASH: code:8084, data:376, headers:8944   free for files:2014212
teensy_size:    RAM1: variables:8896, code:6384, padding:26384   free for local variables:482624
teensy_size:    RAM2: variables:12384  free for malloc/new:511904
============================================= [SUCCESS] Took 4.40 seconds =============================================
```

For others, as explained in https://community.platformio.org/t/platformio-teensy-project-blink-cc-is-included-and-should-not-be/25070, the build fails 

```
.pio/build/teensy41/libFrameworkArduino.a(Blink.cc.o): In function setup': Blink.cc:(.text.setup+0x0): multiple definition of setup’
.pio/build/teensy41/src/main.cpp.o:main.cpp:(.text.setup+0x0): first defined here
/home/greaterebiz/.platformio/packages/toolchain-gccarmnoneeabi/bin/…/lib/gcc/arm-none-eabi/5.4.1/…/…/…/…/arm-none-eabi/bin/ld: Disabling relaxation: it will not work with multiple definitions
```
(conflict with `src/main.cpp`).

* --> This `Blink.cc` shouldn't be in the package
* on some (?) machines it causes a build failure because the user can't implement `setup()` and `loop()` anymore
* both working and non-working build logs show platform-teensy version 4.14.0 being used

valeros, 2021-12-22:
Hi @maxgerhardt thanks for reporting. I fixed the problem by excluding this file from the build process, so it should be resolved in the dev branch.

closed 2021-12-22 by valeros

jeffkaufman, 2024-05-10:
I was getting this intermittently with Arduino IDE 2.3.2 on Mac.  Commenting out the body of `~/Library/Arduino15/packages/teensy/hardware/avr/1.59.0/cores/teensy4/Blink.cc` resolved it for me.

commit 2f3c4230 (2021-12-22): Exclude irrelevant source file from teeny4 Arduino core // Resolve #82