platformio/platform-teensy #84: Please update, Teensyduino 1.56 has been released

skunktrading, 2021-12-23:
Teensyduino 1.56 has been released

PaulStoffregen, 2021-12-23:
This time I remembered to tag a release on Github.  Hope that helps?

https://github.com/PaulStoffregen/cores/releases/tag/1.56

PaulStoffregen, 2021-12-23:
Please also make sure PlatformIO users with Teensy get this copy of FastLED when using Teensy.

https://github.com/PaulStoffregen/FastLED

2 important fixes for Teensy users haven't been merged to upstream FastLED yet.  We've had several reports of these FastLED bugs on the forum, including some reports from PlatformIO users.

closed 2021-12-27 by valeros

valeros, 2021-12-27:
HI @skunktrading ! The Teensyduino package has been updated to the latest 1.56. You can try it out using the upstream version of the platform:
```
[env:teensy40]
platform = https://github.com/platformio/platform-teensy.git
framework = arduino
board = teensy40 
```
A new stable release of the `teensy` platform is planned to published next month.

commit acbb405f (2021-12-27): Update Teensyduino to v1.56 // Resolve #84