platformio/platform-teensy #85: Test fails to start with build option -ffast-math

Foaly, 2022-01-10:
Hey there!

I am observing some weird behaviour that looks like a bug to me.
I have a `main.cpp` and a `test.cpp` file, which both compile and execute fine on my Teensy 4.1. They both produce the expected output. As soon as I add the line `build_flags = -ffast-math` to my `platformio.ini` my test breaks, while my main file still executes as expected (I can even measure the expected performance increase). Now the way the test breaks is what strikes me as odd. It compiles and uploads just fine. But then the Teensy seems to be just dead. It does not connect to the serial monitor or produce any other sort of output. I can still press the button and set it into bootloader mode and flash a new firmware, but other than that there is no response.

I know `-ffast-math` is considered an unsafe option, but what makes me wonder is that the behaviour between compile+upload and compile+upload+test is so different. It feel a lot like a weird bug.

Does anybody have an idea what could be causing this?

-------------

I did some more testing and here are my findings.
For my setup I am running a minimal example like this:
`test/test.cpp`
```cpp
#include <Arduino.h>
void setup() {
    Serial.println("Hello!");
}
void loop() {
    Serial.println("Weeeeeeehhh");
}
```
`platformio.ini`
```ini
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
upload_protocol = teensy-cli
```
As discribed the above example works fine. If I add the line `build_flags = -ffast-math` to the bottom of the `platformio.ini` the described breakage occures when running the test using `pio test -v`.
Since according to the [GCC Optimize Options](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html) `-ffast-math` is a combination of flags I tested the individual flags and this is the result:
| Flag  | Result  |
|---|---|
| -fno-math-errno  | causes the error |
| -funsafe-math-optimizations  | causes the error   |
| -ffinite-math-only  |  causes the error  |
| -fno-rounding-math | good |
| -fno-signaling-nans | good |
| -fcx-limited-range | good |
| -fexcess-precision=fast | good |

Thanks a lot to maxgerhardt who helped me on the platformio community forum. For more info see this thread: https://community.platformio.org/t/test-fails-to-start-with-build-option-ffast-math/25342

PaulStoffregen, 2022-01-12:
As a quick test, I tried -ffast-math on Arduino 1.8.19 with Teensduino 1.56.  I edited boards.txt, changing this line:

    teensy41.build.flags.cpp=-std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing

to this:

    teensy41.build.flags.cpp=-std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -ffast-math

Quick upload and opening the Arduino Serial Monitor seems like it's working fine.

[image]

This is the code I tested:

[image]

I can't say why it's not working for you.  But hopefully this test with these screenshots is detailed and specific enough for you or anyone to reproduce and see for yourself -ffast-math working.

PaulStoffregen, 2022-01-12:
Quick note - you'll see the screenshot shows I used CPU speed 528 MHz.  I also tested at 600 MHz.  Then I changed the CPU speed and tested again, just to force the Arduino IDE to recompile all files.  All other settings were at their default values.

Foaly, 2022-01-14:
Hey Paul!
First of all thank you very much for taking the time and trying it out for yourself! 🙂 

I think you might have misunderstood the problem I was describing. As I write above I was also able to compile and upload code that uses the `-ffast-math` flag. I was even able to measure the expected performance increase. The problem I am having is if I use `pio test` instead of `platformio run --target upload` using the exact same code my Teensy hangs in a non-responsive state. I am suspecting that this is a platformio issue more than an issue with the Teensy.