platformio/platform-teensy #89: Errors comping with EEPROM.h on TD v1.57

david-res, 2022-07-30:
I just updated to TD 1.57 on PIO

When compiling I get the following errors:
`Compiling .pio/build/teensymm/src/memory.cpp.o
In file included from src/memory.cpp:2:0:
/Users/XXXXX/.platformio/packages/framework-arduinoteensy/libraries/EEPROM/EEPROM.h:171:14: error: 'String' does not name a type
 inline const String &EEPROMClass::put(int idx, const String &s)
              ^
/Users/XXXXX/.platformio/packages/framework-arduinoteensy/libraries/EEPROM/EEPROM.h:191:8: error: 'String' does not name a type
 inline String &EEPROMClass::get(int idx, String &s){
        ^`
Never had them on 1.56

valeros, 2022-07-30:
Hi @david-res ! Any chance you could try your project in the Arduino IDE?

david-res, 2022-07-30:
@valeros Unfortunately not :( the project folder structure doesn’t work in Arduino IDE

I might try a simple sketch that writes to EEPROM and see if I get the same error.

BTW should eeprom.h not have Arduino.h included?

david-res, 2022-07-30:
@valeros I included Arduino.h and it now compiles
`#include <inttypes.h>
#include <avr/eeprom.h>
#include <avr/io.h>
#include <Arduino.h>`

I will do a test sketch tomorrow and see if it also happens in Arduino IDE

valeros, 2022-07-30:
1.57 brought some significant changes to the EEPROM library, it's now a header-only library and the Arduino header is no longer included by default. I expect it to work in the Arduino IDE because they use the `ino` format which implies adding the Arduino header behind the scenes.

I guess you just need to add `#include <Arduino.h>` to `src/memory.cpp` just above the line where you include `EEPROM.h`.

david-res, 2022-07-31:
@valeros thanks! All sorted and working again. You can close the issue

closed 2022-07-31 by valeros