platformio/platform-teensy #90: Build warning using TD1.57

skunktrading, 2022-07-31:
Hi. Since upgrading (Thanks very much btw) I get the following warning during LDF when building:
`Warning! Ignoring broken library manifest in /home/skunk/.platformio/packages/framework-arduinoteensy/libraries/FreqCount
`
Other than that my project builds and runs just fine with 1.57

david-res, 2022-07-31:
I have the same warning.
Added the missing comma and it fixed the issue

skunktrading, 2022-07-31:
Thanks for confirming. I'm not sure whether this is Paul's bug or if that that file is generated by some pio scripting.

PaulStoffregen, 2022-08-01:
Fixed https://github.com/PaulStoffregen/FreqCount/commit/8e993136939a15542611c790e2b76eb9bb6d165a

valeros, 2022-08-01:
Thanks for reporting, please run `pio platform update teensy` in the IDE terminal to get the latest package with the bugfix.

closed 2022-08-01 by valeros