platformio/platform-teensy #92: tool-teensy not compatible with linux_aarch64

clintlombard, 2022-08-10:
Running into a strange issue. When using the teensy platform it installs `tool-teensy == 1.21.0`, which is 4 years old. This then fails to upload with the following error:
```
sh: 1: teensy_size: not found
```
If I try manually install a newer version of `tool-teensy` I then get the following error:
```
Error: Could not find the package with 'platformio/tool-teensy @ ~1.157.0' requirements for your system 'linux_aarch64'
```

This used to work before, but something seems to be broken with how platformio is resolving this dependency.

PaulStoffregen, 2022-08-10:
Not sure if this helps, but now you can get teensy_size and other tools without having to run the installer and dig around inside a copy of Arduino to find the files.  They're now also published as package files for Arduino IDE 2.0-rc9.

Here's a direct link to the 1.57 tools for Linux aarch64.

https://www.pjrc.com/teensy/td_157/teensy-tools-linuxaarch64.tar.bz2

clintlombard, 2022-08-11:
Thanks @PaulStoffregen that helps a lot. The package is still broken on platformio, but adding this to the `PATH` fixes my project's build.

valeros, 2022-08-12:
Hi @clintlombard, thanks for reporting. The package for Arm64 architecture has been uploaded.

closed 2022-08-12 by valeros