platformio/platform-teensy #94: Compiler error with c++ templates using teensy platform only

ktgilliam, 2022-10-17:
I can compile this code using gcc or the native platform in platformio, but I am getting an error when I build for the Teensy4.1:

```
template <typename T>
inline T saturate(T val, T lower, T upper)
{
   T val1 = val > lower ? val : lower;
   T val2 = val1 < upper ? val1 : upper;
   return val2;
}

int main()
{
    double v = 1.234;
    saturate(v, 0.0, 90.0);
    return 0;
}
```

The error I get is:
`src/exec/MainExec.cpp: In function 'int main()':
src/exec/MainExec.cpp:12:26: error: no matching function for call to 'saturate(double&, float, float)'
     saturate(v, 0.0, 90.0);`

Is this a bug with the teensy platform or with the compiler?

PaulStoffregen, 2022-10-19:
I tried with Arduino IDE and it seems to work.

[image]

PaulStoffregen, 2022-10-19:
However, it does fail if I use Teensy 3.2, where code is compiled with "-fsingle-precision-constant" option.  This is the expected result with -fsingle-precision-constant.

Perhaps PlatformIO is incorrectly using -fsingle-precision-constant with Teensy 4.1?  It is only supposed to be used with the older boards.

blackketter, 2022-10-19:
Yup, it seems that -fsingle-precision-constant is used with all boards on platformio.  I'll file a bug.

commit 4af2ef03 (2022-10-24): Remove unnecessary "-fsingle-precision-constant" flag for Teensy4 core

Resolves #94, resolves #95, resolves #96
closed 2022-10-24 by valeros