platformio/platform-teensy #97: implement teensy 1.58beta2

bernhardkurz, 2022-10-23:
@ivankravets : how difficult would it be to implement a beta version of teensyduino? i tried to create a local source with 1.58beta2 files from teensy, but i get linker errors.
i also "hijacked" a thread at [platform.io](https://community.platformio.org/t/teensy-using-a-beta-version-of-the-platform-package/21381/14) where this was discussed for earlier beta implementation. below is a copy of my posting there

this was my steps:

1. install arduino 1.8
2. install teensyduino 1.58beta (as inspired by [joepasquariello](https://forum.pjrc.com/members/45436-joepasquariello)) → just to get the files
3. in [.platformio\packages] i create folder [framework-arduinoteensy1158] with files from arduino installation
4. in [.platformio\packages] i create folder [toolchain-gccarmnoneeabi11.3.rel1] and put the files from [armDevelopers gcc11.3rel1](https://developer.arm.com/-/media/Files/downloads/gnu/11.3.rel1/binrel/arm-gnu-toolchain-11.3.rel1-mingw-w64-i686-arm-none-eabi.zip?rev=82f0e95b5a7740d9899e6ce375df8106&hash=279E24D725F493C4E3B90C8B2A159813A91CD29C)
5. added reworked files package.json and .piopm to the folders
6. modified platform.ini to use the specific folders and force -std=gnu++17

```
[env:teensy41]
;platform = https://github.com/platformio/platform-teensy.git
;platform = teensy
board = teensy41
framework = arduino
platform_packages =
	framework-arduinoteensy@file://C:/Users/Admin/.platformio/packages/framework-arduinoteensy1158
	toolchain-gccarmnoneeabi@file://C:/Users/Admin/.platformio/packages/toolchain-gccarmnoneeabi11.3.rel1
lib_compat_mode = strict
lib_ldf_mode = chain+

build_flags =
    -std=gnu++17
	;-fexceptions
build_unflags =
    -std=gnu++14
	;-larm_cortexM7lfsp_math
```

if it is not possible to create a git source, maybe you could guide me to successfully implement manually?

the exit fault with my setup is this (the outputs with warnings is quite long, so i truncated the output to what i think is the problem):

```
Linking .pio\build\teensy41\firmware.elf
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.startup+0x0): relocation truncated to fit: R_ARM_PREL31 against `.startup'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.startup+0x8): relocation truncated to fit: R_ARM_PREL31 against `.startup'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.startup+0x10): relocation truncated to fit: R_ARM_PREL31 against `.startup'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x0): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'        
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x8): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'        
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x10): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'       
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x18): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'       
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x20): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'       
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x28): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'       
.pio\build\teensy41\libFrameworkArduino.a(analog.c.o):(.ARM.exidx.flashmem+0x0): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(tempmon.c.o):(.ARM.exidx.flashmem+0x0): additional relocation overflows omitted from the output
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\teensy41\firmware.elf] Error 1
============================================================= [FAILED] Took 54.29 seconds =============================================================
 *  Der Terminalprozess "C:\Users\Admin\.platformio\penv\Scripts\platformio.exe 'run'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```
any helpf is apreciated, thank you in advance

PaulStoffregen, 2022-10-23:
.ARM.exidx "relocation truncated to fit: R_ARM_PREL31" probably means you are using the new toolchain (for 1.58-beta2) with the old linker scripts (from 1.57).

PaulStoffregen, 2022-10-23:
This is the 1.58-beta2 change you're probably missing:

https://github.com/PaulStoffregen/cores/commit/d496c4579e86821f6a8dbeb2f0a7bbbd5288331b

Further explanation here:

https://forum.pjrc.com/threads/71048-Teensyduino-1-58-Beta-1-(updated-toolchain-trial)?p=312373&viewfull=1#post312373

bernhardkurz, 2022-10-23:
Dear Paul,

thank you for investigating.
i changed those 3 files as suggested it shows me this error 

```
c:/users/admin/.platformio/packages/toolchain-gccarmnoneeabi11.3.rel1/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -larm_cortexM7lfsp_math: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\teensy41\firmware.elf] Error 1
```

so in [platform.ini] add

```
build_unflags = -larm_cortexM7lfsp_math
```

now i am able to build and upload (program is working as expected) which is a great milestone in this issue ;) so thank you very much at this place.

unfortunately when i include "struct_mapping" (actually the reason for all the effort) it throws following error

```
.pio\libdeps\teensy41\struct_mapping\include/struct_mapping/member_string.h:41:25: error: exception handling disabled, use '-fexceptions' to enable
   41 |                         throw StructMappingException("MemberString not set for member: " + name);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\teensy41\libdc8\jsonMapping\jsonDrv.cpp.o] Error 1
```

finally when i add

```
build_flags = -fexceptions
```

i am back in death valley,
maybe there is something i still miss? any hint is much appreciated

```
Linking .pio\build\teensy41\firmware.elf
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.startup+0x0): relocation truncated to fit: R_ARM_PREL31 against `.startup'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.startup+0x8): relocation truncated to fit: R_ARM_PREL31 against `.startup'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.startup+0x10): relocation truncated to fit: R_ARM_PREL31 against `.startup'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x0): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x8): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x10): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x18): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x20): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(startup.c.o):(.ARM.exidx.flashmem+0x28): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(analog.c.o):(.ARM.exidx.flashmem+0x0): relocation truncated to fit: R_ARM_PREL31 against `.flashmem'
.pio\build\teensy41\libFrameworkArduino.a(tempmon.c.o):(.ARM.exidx.flashmem+0x0): additional relocation overflows omitted from the output
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\teensy41\firmware.elf] Error 1
======================================================================================= [FAILED] Took 117.42 seconds =======================================================================================
 *  Der Terminalprozess "C:\Users\Admin\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
 * ```

ivankravets, 2023-03-31:
Duplicate of https://github.com/platformio/platform-teensy/issues/101

closed 2023-03-31 by ivankravets