platformio/platform-teensy #101: Add support for Teensyduino 1.58

VictorFSantolim, 2023-03-31:
Teensyduino V1.58 just got released today.
[PJRC forum link](https://forum.pjrc.com/threads/72509-Teensyduino-1-58-Released)
[Download link](https://www.pjrc.com/teensy/td_download.html)
@valeros I see that you updated platform-teensy with previous releases from PJRC. Do you plan to integrate this new release into PlatformIO soon?

Thank you for all the support!

PaulStoffregen, 2023-03-31:
This time I remembered to tag a release on the core library.  Hopefully it helps?

https://github.com/PaulStoffregen/cores/releases/tag/1.58

We've updated to a newer toolchain.  Here are direct links to the toolchain downloads (same as inside the JSON for Arduino IDE 2.0.x)

https://www.pjrc.com/teensy/td_158/teensy-compile-linux32.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-compile-linux64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-compile-linuxaarch64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-compile-linuxarm.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-compile-macos.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-compile-windows.tar.bz2

valeros, 2023-04-01:
Hi @PaulStoffregen ! Many thanks for the toolchains, it will help a lot.

I'm wondering if you have similar packages but for your additional tools (teensy_size, teensy_post_compile, etc.)?

PaulStoffregen, 2023-04-01:
Yes.  They're all in the JSON.  Here's the links.

Tools:

https://www.pjrc.com/teensy/td_158/teensy-tools-linux32.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-tools-linux64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-tools-linuxaarch64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-tools-linuxarm.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-tools-macos.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-tools-windows.tar.bz2

Device Discovery:

https://www.pjrc.com/teensy/td_158/teensy-discovery-linux32.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-discovery-linux64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-discovery-linuxaarch64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-discovery-linuxarm.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-discovery-macos.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-discovery-windows.tar.bz2

Serial Monitor:

https://www.pjrc.com/teensy/td_158/teensy-monitor-linux32.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-monitor-linux64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-monitor-linuxaarch64.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-monitor-linuxarm.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-monitor-macos.tar.bz2

https://www.pjrc.com/teensy/td_158/teensy-monitor-windows.tar.bz2

PaulStoffregen, 2023-04-01:
This is the link for the main JSON file, which gives the links and info about everything.

https://www.pjrc.com/teensy/package_teensy_index.json

valeros, 2023-04-03:
Updated to 1.58, anyone who wants to test it out before a new release, please use the upstream version of the platform (a Git client is required):
```ini
[env:teensy41]
platform = https://github.com/platformio/platform-teensy.git
framework = arduino
board = teensy41
```

commit cc5d5c68 (2023-04-06): Update Teensyduino to v1.58 // Resolves #101

Also brings updated toolchain packages v11.3.1