platformio/platform-teensy #102: PlatformIO 4.18.0 Teensy package name change breaks ability to compile with other toolchain/GCC versions

egonbeermat, 2023-04-19:
In 4.17.0, I could override the default toolchain to use GCC 9.3.1 with this statement in platformIO.ini:

platform_packages = toolchain-gccarmnoneeabi@1.90301.200702

Since upgrading to 4.18.0, it appears that it is now referencing the package toolchain-gccarmnoneeabi-teensy, the above statement no longer overrides, and there is no other version of the new package except for 11.x. 

How can I achieve the use of GCC 9.3.1 with 4.18.0?

ivankravets, 2023-04-24:
Teensy depends on the specific version of the toolchain that contains extra libraries. Also, teensy's toolchain is incomplete and you can't use it with non-teensy projects. See the reported issue about the buggy teensy toolchain https://github.com/platformio/platformio-core/issues/4590

So, we decided to fix teensy dev-platform to the same toolchain that is used in the Teensyduino. We don't support other toolchains and don't provide any help when people try to use them.

cc: @PaulStoffregen

PaulStoffregen, 2023-04-24:
Yes, confirm, the toolchain for Teensy lacks many of the multilib variants not needed for any Teensy boards.  It was never meant to be distributed as a general purpose toolchain for all boards.  Deleting that unnecessary (for Teensy) stuff drastically reduces the download size.

I have an issue report pending where the default build of newlib might not be quite right for the float registers on Teensy 4.x.  If that turns out to be true, we will rebuild newlib with different settings.  Again, everyone should understand this build of the toolchain is really only meant to be used with Teensy.

PaulStoffregen, 2023-04-24:
If anyone is interested in the original builds of the arm-none-eabi toolchain with all multilib variants intact and without the extra libs and AVR stuff added, I would be happy to provide them.  The files are huge, approx 700MB per platform, rather than the 120MB files we distribute for Teensy's toolchain.  Just say ask and I will put them temporarily on the PJRC server.  Please understand this is only so you can download them and host them on your own site.  PJRC isn't going to host those 6 huge files, as we're already sustaining quite a lot of website bandwidth and recently had to upgrade to a server allowing more monthly data transfer.

ivankravets, 2023-04-25:
@PaulStoffregen , thanks for the provided information. We will use it as the source if people will have similar questions.

Regarding the big files. You can use Github Releases and host files here. It should save your server bandwidth.

PaulStoffregen, 2023-04-25:
Ok, files are here:

https://github.com/PaulStoffregen/ARM_Toolchain_2022q3_Source/releases/tag/release

egonbeermat, 2023-04-25:
I'm not sure this helps, or responds to my original question? What I need would be to have a toolchain-gccarmnoneeabi-teensy@1.90301.200702 available, not only the 11.x version, or information on how I would change the reference from toolchain-gccarmnoneeabl-teensy back to how it was in prior versions of PlatformIO, so I can override as before?

It's not a given that that latest compiler produces the fastest code, and the output from 11.x in one critical area is around 10% slower than what I achieved using 9.3.1, taking me code below a threshold it needs to achieve.

PaulStoffregen, 2023-04-25:
I can't help with that request.  Before gcc 11.3.1, the prior toolchain PJRC published was gcc 5.4.1.  The links are in the JSON file for Arduino IDE 2 installs.  Again, those copies are customized for Teensy boards.  The full original code can be found on my github profile, just click repositories and search "toolchain".

ivankravets, 2023-04-25:
See https://github.com/platformio/platform-teensy/releases and https://docs.platformio.org/en/latest/projectconf/sections/env/options/platform/platform.html

```ini
[env:production]
platform = teensy @ 4.17.0
```

PaulStoffregen, 2023-04-25:
@egonbeermat - Would you be willing to share the code you used to measure 10% performance regression?  I'm asking because we stuck with gcc 5.4.1 for so long because many later versions tested significantly slower with -O2 and -O3.  In the move to finally update to 11.3.1, quite a lot of benchmarking was done.  We found it generated slightly faster code in almost all cases.  Compiled code size did increase with -O2 and -O3 in many cases, often by 5% to 12%.  So I'm really curious to see the specific performance test you did that showed slower generated code?